tier4/CalibrationTools #19: Extrinsic ground-plane calibration tool

NilaySener, 2022-08-12:
Hi, I'm trying to use your Extrinsic ground-plane  calibration tool for my vehicle by following the instructions in the [link](https://github.com/tier4/CalibrationTools/blob/tier4/universe/sensor/docs/how_to_extrinsic_ground_plane.md),  but I have some questions about the integration in the launch parameters.
 Can you provide detailed information about the file locations of my sensor_model and vehicle_model packages and the properties of the topics that should be in my rosbag file? Also, is it necessary to have a map or something among the parameters, or are the sensor and vehicle parameters sufficient for the tool?

knzo25, 2022-08-12:
Hi !
Thank you for your interest in our package.

We only require point clouds in the rosbag itself, but we need the TF tree, so in the case of autoware, we need to run the logging simulator (which is set by default in https://github.com/tier4/CalibrationTools/blob/tier4/universe/sensor/extrinsic_calibration_manager/launch/calibration.launch.xml)

In term of parameters, strictly speaking, this particular tool does not require any additional information (e.g., maps) or vehicle parameters (e.g, dimensions) since the algorithm is quite simple (RANSAC + filtering). The additional overhead of parameters (i.e., sensor_model, vehicle_model, and vehicle_id) are only to conform with the API and integrate it with the pipeline.

In the documentation the example to run the tool is:

```
ros2 launch extrinsic_calibration_manager calibration.launch.xml \
  mode:=ground_plane sensor_model:=aip_x1 vehicle_model:=lexus vehicle_id:=my_awesome_vehicle
```

where
 - `sensor_model`: This is a key aspect of autoware and its both used to run the `logging_simulator` and the calibration tools. For example, in https://github.com/tier4/CalibrationTools/tree/tier4/universe/sensor/extrinsic_calibration_manager/launch, you can see that for every sensor model (`aip_x1` in the example), we implement different launchers to account for the type/number of sensors to calibrate.
 - `vehicle_model`: We use `vehicle_model` to find the `$(find-pkg-share $(var vehicle_model)_description)` package. For example, if you set `vehicle_model` to `my_vehicle_model` it would look for `my_vehicle_model_description`, which should be located in `autoware/src/description/vehicle/my_vehicle_model_description` 
 - `my_awesome_vehicle:`  Is the name of a particular vehicle and we use it to find its parameters in the form: `$(find-pkg-share individual_params)/config/$(var vehicle_id)/$(var sensor_model)/sensor_kit_calibration.yaml`. The `individual_params` package should be in `autoware/src/autoware/individual_params`.

As you can see the current way to launch the tools is strongly linked to `autoware` (the tools' implementation is not). It had been previously discussed that we could also provide an `autoware`-free alternative to use this repository, but only if there was demand for it, so if you need it please let us know !

knzo25, 2022-09-01:
@NilaySener any updates?

knzo25, 2022-09-08:
I am closing this issue due to inactivity.

closed 2022-09-08 by knzo25