tier4/CalibrationTools #84: Dimensions of boards

bariskapaligoz, 2023-02-09:
Hi, i want to test your tag based calibration tool. I have a few questions about dimensions of your boards.

[image]

I am wondering lengths of fiducial marker(Yellow square), black edge and outermost white edge. The reason I ask is because I want the board we will use to be compatible with the default parameters that you define in both apriltag and lidartag.

Thank you in advance.

knzo25, 2023-02-09:
Hi, our boards are 0.8m side-to-side, making each "cell" you see in the image to be 10cm x 10cm.
Let us know if there is anything we can help you with !

bariskapaligoz, 2023-02-09:
Thank you very much for the quick help!

closed 2023-02-09 by bariskapaligoz