tier4/CalibrationTools #89: Failed to fit a plane model to the cluster

bariskapaligoz, 2023-02-16:
Hello, my goal is to do label-based calibration with a pair of Ouster OS1 128 and Lucid Camera. We prepared boards as a your dimensions and collected bag.
[image]

The command I tried for calibration:
`ros2 launch extrinsic_calibration_manager calibration.launch.xml   mode:=tag_based sensor_model:=aip_x2 vehicle_model:=gsm8 vehicle_id:=ps1 camera_name:=camera0 logging_simulator:=false`

I have been change `lidar_model` to "velodyne_vls128" cause of I have 128 beam lidar. 

As a last I got  "Failed to fit a plane model to the cluster" error from lidartag. Where could I have made a mistake?

knzo25, 2023-02-16:
Hi!
Thanks for your interest in our tools.

Although we do not have in our hands an Ouster OS1, a while ago, we could run the camera-lidar calibration with a [OS2](https://github.com/tier4/CalibrationTools/issues/43).

The error itself suggests that outliers made their way into the clusters (e.g., the tripods or the person behind the tag).
The lidartag package is a tricky one to debug since the number of clusters is quite big, but the markers from the topics `colored_cluster_buff`, `cluster_buff_points_size_markers`, and `cluster_buff_index_number_markers` provide useful information to understand what is going inside the detection pipeline.

That being said, we are quite interested in expanding the range of sensors we give support to, so we could analyze your data and get it working for you ;) (that is, if you and your organization are ok with it). 

Good luck with your projects and drop us a line in case you need anything from us !

bariskapaligoz, 2023-02-16:
Thanks for quickly reply.  
Yes, I think I understood better what you said when I looked at it from the rviz. I will try with filtered pointcloud on z-axis so that the tripods are not visible.

[image]

knzo25, 2023-02-16:
Two additional points.

1.  The parameter that regulates the clustering is `<arg name="linkage_tunable" value="0.2" />`. If you reduce it, you can limit the merging of clusters.
2. I am not sure anymore since I don't have an OS1, but if the ring information is not correctly included in the pointcloud field, you may need to use the Lidartag version of this PR: https://github.com/tier4/LiDARTag/pull/15

bariskapaligoz, 2023-02-17:
[image]

Firstly, I tried to use filtered pointcloud on z-axis. At least the cluster of 2 boards looks fine. You can see above visualize result  of filtering. But unfortunately nothing has changed.

After than i changed lidartag package branch to  as a `feature/add_more_lidar_support`. But again nothing has changed.

Finally, I compared the data in my bag with the data in the bag I downloaded from lidartag. In the images below you can see the visualization of both pointclouds according to the ring information. I couldnt see any issue.

### My pointcloud
[image]

###  Pointcloud of the bag downloaded from Lidartag Repository
[image]

Honestly, I still don't quite understand what the problem is. Do you have any ideas about this?

knzo25, 2023-02-17:
I see, the clusters are perfectly detected yet they do not produce a detection.

Lidartag publishes several visualization makers (among those, I recommend seing `cluster_buff_points_size_markers` and `cluster_buff_index_number_markers`). The detection pipeline is linear and when a cluster is discarded, an error ID is assigned following the order presented in this [link](https://github.com/tier4/LiDARTag/blob/humble/include/lidartag/types.hpp).

If you know what error ID those cluster have, you can check which filter is failing.

(Just in case, let me reiterate the previous offer of  checking the data from our side. We would require either a bag or a pcd which maintains all the required fields: xyz, intensity, and rings)

bariskapaligoz, 2023-02-17:
Thank you for your help. Okay then here is the link so you can download the bag I used. 

`https://netorgft4229778-my.sharepoint.com/:f:/g/personal/baris_adastec_com/ElNgLo-3xkJKgpbEEjxkE-QB49e-fKl-oBGol3xgm-mLwg?e=PUO3Qv`

knzo25, 2023-02-17:
Thanks for sharing !
I just confirmed the bag and using the following launch file

```
<launch>

  <arg name="pointcloud_topic" default="/pointcloud_raw" />
  <arg name="use_intensity_channel" default="true" />
  
  <include file="$(find-pkg-share lidartag)/launch/lidartag_master.launch.xml" >

    <arg name="pointcloud_topic" value="$(var pointcloud_topic)" />
    <arg name="use_intensity_channel" value="$(var use_intensity_channel)" />
    <arg name="beam_number" value="128" />
    <!--arg name="payload_intensity_threshold" value="0.001"/-->
    <arg name="use_organized_pointcloud" value="true"/>

    <!-- list for marker sizes -->
    <arg name="tag_size_list" value="[0.8]" />

    <!-- Tunable for PoI clustering -->
    <arg name="nearby_factor" value="2.0" />
    <arg name="linkage_tunable" value="0.22" />

    <!-- SQRT2 / 4 ==  0.35355339059-->
    <arg name="clearance" value="0.35355339059" />

    <!-- Tunable for cluster validation -->
    <arg name="max_outlier_ratio" value="0.25" />

    <arg name="cluster_min_index" value="0" />
    <arg name="cluster_max_index" value="10000" />
    <arg name="cluster_min_points_size" value="0" />
    <arg name="cluster_max_points_size" value="10000" />
    <arg name="pcl_visualize_cluster" value="false" />

  </include>

</launch>
```

[image]

Can you confirm to us if it works now on your end?
`ros2 launch lidartag lidartag_os2_128.launch.xml pointcloud_topic:=/lidar/parent/points_raw`

bariskapaligoz, 2023-02-17:
Yes it definitely worked!
[image]

Thanks again for help.

knzo25, 2023-02-17:
Don't forget to mark the issue as closed ;)

closed 2023-02-17 by bariskapaligoz