tier4/CalibrationTools #91: Waiting for service:

bariskapaligoz, 2023-02-24:
Hello, after starting the calibration tool, as you can see below, the waiting for service information is constantly coming and stays like this.
`[extrinsic_calibration_manager-3] [INFO] [1677242419.329916718] [dash_center.extrinsic_calibration_manager]: Waiting for service: dash_center
`
When I look at Apriltag and Lidartag detection output issues they both show up. Lidartag detection may have trouble recognizing one of its boards. Could this have an effect?

I did not fully understand why it remained like this, what do you think could be missing or wrong?

knzo25, 2023-02-24:
Hi, the `Waiting for service` message is not related with the state of the calibrator.

What is happening is probably that due a mismatch between the server client/servers' names. You can check them via `ros2 service list` (make sure the tool and the calibration manager match)

With respect to the missing detection, in theory you can perform PnP with just one board, but unless the intrinsics are perfect (they are not) it will not give good enough results. With two boards it already works well more or less (again, depending on the quality of the intrinsics), and with 3 more you can see that the calibration converges rapidly.

In the bag you provided the other day, the three tags are static, but in order to collect more samples (pairs of detections) what we usually do is move them from a location to another every 10 seconds (that way we only use one tag)

Let me know if you have any more doubts !

knzo25, 2023-03-26:
@bariskapaligoz 

Given that is issue has been inactive for about a month, I will be closing it in a week or so

knzo25, 2023-04-02:
Closing due to inactivity

closed 2023-04-02 by knzo25