tier4/CalibrationTools #93: finding X, Y and Yaw for LiDAR base link

ajay1606, 2023-03-09:
Thank you so much for very useful contribution.

After going base lidar calibration, we are able to get **Z** and **roll, pitch** angle for LiDAR base_link. 

I just wonder, why its not possible to automate **X, Y** and **YAW**, which also most important params to find. 

Similar surface plane detection, can we detect vertical planes (ex Wall structrue, wide marker board placed front of vehicle) to estimate X, Y, Yaw angle. 

Initial parameters can be vehicle parked location to marker board/ wall location. 

Appreciate your opinion on possibility and challenges involved

knzo25, 2023-03-09:
Hi,

Going by the definition of `base_link` as a frame that lies on the ground in the middle of the rear wheels, you can understand why our previous method only goes as far as `roll`, `pitch`, `z` (the ground plane).

If you use wall structure or a wide marker board, you can indeed get the `y` and `yaw` components if you really make an effort, and align things correctly. However, and this is my opinion, this involves several critical measurements and the end result ends up having either not a great result or getting to close to manual calibration (which in our experience is quite time consuming to get it completely right).

Even so, getting the `x` component would just become manual calibration at this point.

If you want to have a fully automated - full pose base-lidar calibration, we are making one in [this](https://github.com/tier4/CalibrationTools/tree/feature/tag_based_base_calibrator/sensor) branch. Currently it achieves 1cm for the xyz and 0.5 degrees of normal difference with our manual calibration (the difference is in part due to our tags nor being very good at the moment).

The method itself relies on several markers (including some on the wheel) and an external camera (we use a Nikon DSLR) and performs bundle adjustment between the Nikon and the lidar detections. As an additional bonus, you can even calibrate multiple lidars and cameras plus the base link all in one.

We are planning on taking some data for the OSS release and documentation at the end of the month. If you are interested let me know and I will keep you updated !

ajay1606, 2023-03-09:
@knzo25 Thank you so much, It is very interesting work, Calibrating multiple Lidar, Camera, and base link parameters in single calibration platform.  

I am eagerly looking forward to check out your upcoming OSS release, kindly consider to inform once it released.

Regards,
Ajay

closed 2023-03-09 by ajay1606

knzo25, 2023-08-02:
@ajay1606 

Securing the work hours for a review of the beforementioned tool took way longer than anyone expected.
However, it was merged in https://github.com/tier4/CalibrationTools/commit/128d68672f4a180c8c3dc1351bda3dbef998a5db

ajay1606, 2023-08-02:
@knzo25 Excellent, For sure we will give it a try, Thanks a lot.