tier4/CalibrationTools #96: How to input object and image points to Extrinsic Interactive Calibration (camera-lidar)

ajay1606, 2023-03-26:
Hello,

Thank you so much for the very useful and well documented to repository.
Currently am trying to calibrate Lidar and Camera using Extrinsic Interactive Calibration (camera-lidar) method.

But i didnt understand, how to input object and Image points to estimation 3D projection matrix. Currently, i have recorded bag file
with LiDAR and Camera data. 

Image: /blackfly_0/image_raw
Point cloud: /points_raw 

So is there options to click points from LiDAR and mage data, for Object and Image coordinates ?

`ros2 launch extrinsic_calibration_manager calibration.launch.xml   mode:=interactive sensor_model:=aip_x2 vehicle_model:=gsm8 vehicle_id:=ps1 camera_name:=camera1 logging_simulator:=False interactive_use_concatenated:=False`

Should i publish static TF ? as a initial guess parameters ?

knzo25, 2023-03-26:
Dear @ajay1606 
I have difficulties understanding your problem.

The tool itself collects pairs of object-image points and uses standard PNP.
Pointcloud points are collected using Rviz, whereas image points are collected by clicking in the UI

ajay1606, 2023-03-27:
@knzo25 Thank you so much for your response.

Point cloud points are collected using RVIZ means, user has to select specific feature points right ! In my case its not working.

currently am launching module with following command:
`ros2 launch extrinsic_calibration_manager calibration.launch.xml   mode:=interactive sensor_model:=aip_x2 vehicle_model:=gsm8 vehicle_id:=ps1 camera_name:=camera1 logging_simulator:=False interactive_use_concatenated:=False`

And am able to visualize the Image and Point cloud data like below;

[image]

And am trying to choose point from LiDAR and Camera using Publish point, but clicked points not published at `/clicked_points` topic. 

Also unable to visualized image in the window

[image]

Please any suggestions please.

knzo25, 2023-03-28:
Hi, the most probable reason is because the topics don't match.
See the output of `ros2 node info` to check that one.

The other possible reason, is that the data is not synchronized, in which case you would do well to see the contents is the camera info, image, and pointcloud's headers

knzo25, 2023-03-29:
Also, is the expected frame ids don't match, the image may also not appear.

ajay1606, 2023-03-29:
@knzo25 I working around with your input and update you once i get expected results. Thank you so much

ajay1606, 2023-04-02:
@knzo25 
After mapping topic names and frame ids, i am able to visualize the Image and point cloud in the tool.

[image]

But in the terminal am getting this message continously.
`[interactive_calibrator-4] The following exception was never retrieved: 'RosInterface' object has no attribute 'lidar_frame'`

Also my TF tree looks below
LiDAR:
[image]

Camera:
[image]

Would please let me know , how can i select Object and Image points from LiDAR and Image visualizer. 

If i clink point in rviz, image visualizer tool dies with following error message.

[image]

Any suggestions please. !

knzo25, 2023-04-02:
Dear @ajay1606,
Although it is true that some checks should be added (the tool itself is subject to a big design change, so I will not do it now), the problem lies with your data, as you can see in the code.

https://github.com/tier4/CalibrationTools/blob/301f383d8e0a4a152900b5d4bda434a58c709ea7/sensor/extrinsic_interactive_calibrator/extrinsic_interactive_calibrator/ros_interface.py#L327

(I.e., you are not receiving the pointcloud)

ajay1606, 2023-04-03:
@knzo25, Thank you for your notice about future change. I will check it again with our data.

ajay1606, 2023-04-04:
@knzo25 

I am currently testing with following static TF,  And input are like this

Lidar: 
**Topic name:** /points_raw 
**Frame_id:** velodyne_top

Camera:
**Topic name:** /out/compressed 
**Frame id:** camera_link

[image]

And checked point cloud in all frame_id, its publishing well. In calibration tool image_viewer, image displaying well. 

But in the following line of code; https://github.com/tier4/CalibrationTools/blob/tier4/universe/sensor/extrinsic_calibration_manager/src/extrinsic_calibration_manager_node.cpp

`target_client.client->wait_for_service(1s)` returns 0, and its continuously logs statement like

` Waiting for service: velodyne_top (calibrationRequestCallback() at`

[image]

Would you please share input, how could i successfully register service ?

knzo25, 2023-04-04:
Hi, I only gave a look to the first part of the post, but be sure to differentiate between the `camera_link` and the `camera_optical_link`. Since the different between the two is a rotation (set by the usual conventions -> hardcoded), we calibrate between the `sensor_kit_base_link` and the `camera_link`

ajay1606, 2023-04-05:
@knzo25 Any comments on " Waiting for service: velodyne_top (calibrationRequestCallback() at" this message in the terminal ?

knzo25, 2023-04-05:
@ajay1606 
It seem that it is, as usual, a mismatch between the expected services and the actual ones. It not something to get rid off, but rather to address

ajay1606, 2023-04-06:
@knzo25 Seems i made it complex with custom sensor TF. May i know your sensor input to this tool (topic name and frame id for LiDAR and Camera). I will change my Sensor inputs accordingly to sync with default params.

knzo25, 2023-04-06:
@ajay1606 

Most topics and frames are here:
`https://github.com/tier4/CalibrationTools/blob/tier4/universe/sensor/extrinsic_calibration_manager/launch/aip_xx1/interactive_sensor_kit.launch.xml`

It will calibrate the tf from the `sensor_kit_base_link` to the `camera_link`.
The lidar's frame can be whatever, as long as it is connected to the `sensor_kit_base_link`.
Also, as I said before, be sure to understand the difference between the `camera_link` and `camera_optical_link` as the image and camera info are in the optical frame, whereas the calibration is done at the `camera_link` frame.

ajay1606, 2023-04-06:
@knzo25 

Herewith am checked like below to confirm sensor TF frames and data. 

[image]

But am continuously getting message in the terminal like below. 

[image]

Would you please clarify why service request is not successful ?

knzo25, 2023-04-06:
About the first error, I would urge you to read how the service naming is implemented. That way you can adjust it to your needs.
https://github.com/tier4/CalibrationTools/blob/301f383d8e0a4a152900b5d4bda434a58c709ea7/sensor/extrinsic_calibration_manager/src/extrinsic_calibration_manager_node.cpp#L130
https://github.com/tier4/CalibrationTools/blob/301f383d8e0a4a152900b5d4bda434a58c709ea7/sensor/extrinsic_calibration_manager/launch/aip_xx1/interactive_sensor_kit.launch.xml#L53

With regards to the second error, I have no idea what is going on. You would need to debug it with prints / breakpoints, etc.

ajay1606, 2023-04-07:
@knzo25 With your inputs, seems LiDAR and Camera inputs working well with proper frame id. I have checked with 
But regarding the second error, in my understanding point clicked in the **rviz** should be received in the following callback.
https://github.com/tier4/CalibrationTools/blob/301f383d8e0a4a152900b5d4bda434a58c709ea7/sensor/extrinsic_interactive_calibrator/extrinsic_interactive_calibrator/ros_interface.py#L400

But above callback not triggers up on point click in the **rviz** and there suddenly image window terminates. 

By the way, any other that i can confirm point clicking routine normal functioning ?

knzo25, 2023-04-07:
If the UI crashes with no error, it is highly possible that you are trying to render something unfeasible.
I recommend you using a debugger like debugpy to check the error. Sadly I do not currently have the leeway to check your data myself.

ajay1606, 2023-04-07:
@knzo25 tHANK For the suggestions ! Sure, I will try with the debugger tool.

https://github.com/tier4/CalibrationTools/blob/301f383d8e0a4a152900b5d4bda434a58c709ea7/sensor/extrinsic_interactive_calibrator/extrinsic_interactive_calibrator/ros_interface.py#L400
Regarding the above callback, subscribing not activating at all` /clicked_point.` But topic echo for the ` /clicked_point` shows result like this. 

[image]

Further, while terminating , Image window there is a error in the terminal and it says. 

```
interactive_calibrator-4] /home/ajay/autoware/build/extrinsic_interactive_calibrator/extrinsic_interactive_calibrator/image_view.py:773: RuntimeWarning: libshiboken: Overflow: Value -37380821088 exceeds limits of type  [signed] "i" (4bytes).
[interactive_calibrator-4]   painter.drawLine(
[interactive_calibrator-4] OverflowError
[interactive_calibrator-4] /home/ajay/autoware/build/extrinsic_interactive_calibrator/extrinsic_interactive_calibrator/image_view.py:773: RuntimeWarning: libshiboken: Overflow: Value 77809694928.97792 exceeds limits of type  [signed] "i" (4bytes).
[interactive_calibrator-4]   painter.drawLine(
[interactive_calibrator-4] OverflowError
[interactive_calibrator-4] /home/ajay/autoware/build/extrinsic_interactive_calibrator/extrinsic_interactive_calibrator/image_view.py:773: RuntimeWarning: libshiboken: Overflow: Value -37380821079.95262 exceeds limits of type  [signed] "i" (4bytes). 
[interactive_calibrator-4]   painter.drawLine(
[interactive_calibrator-4] OverflowError
[interactive_calibrator-4] /home/ajay/autoware/build/extrinsic_interactive_calibrator/extrinsic_interactive_calibrator/image_view.py:773: RuntimeWarning: libshiboken: Overflow: Value 77809694937.14784 exceeds limits of type  [signed] "i" (4bytes).
[interactive_calibrator-4]   painter.drawLine(
[interactive_calibrator-4] QPaintDevice: Cannot destroy paint device that is being painted
[ERROR] [interactive_calibrator-4]: process has died [pid 3536551, exit code -11, cmd '/home/ajay/autoware/install/extrinsic_interactive_calibrator/lib/extrinsic_interactive_calibrator/interactive_calibr
```
I am just trying to figure out, though appears to be in normal topic echo of  ` /clicked_point `, but its values ended as overflow values at `image_view.py`.

But strangely, in very rare trials, point clicks works well and i can click point image viewer as well. But after clicking another point, again overflow error appears.

if you dont mind, will it possible to check my data once ? I will share it, if it is possible for you.

knzo25, 2023-04-07:
With respect to the `clicked_point`, you can check if the node is actually listening to that topic and if the qos match. I actually had to use the interactive tool the other day so it is working (bar any issues with the namings, since I used a custom launcher for a demo).

Overflow values kinda don't make much sense, unless the initial tf is invalid or totally wrong.

As I said before, checking your data is currently impossible for me

ajay1606, 2023-04-07:
@knzo25 Thank you

ajay1606, 2023-04-10:
@knzo25 Thank you so much, after carefully tuning initial params, now able to get Lidar data on image. 

**Interactive UI:**
[image]

**Rviz: **
[image]

Now, able to pick point in rviz, but when i click corresponding point in the image, there were no any response. As It supposed to render a picked point in the image. Am i missing something here ?

knzo25, 2023-04-10:
Maybe the pointcloud is eclipsing the visualization, so in the meantime, please disable  `Render pointcloud`.
Then, after clicking the point in rviz, it should appear in the image as according to the initial/curent tf.

I am pretty sure you are following the tutorial, but I will leave the relevant links
https://github.com/tier4/CalibrationTools/blob/tier4/universe/sensor/docs/how_to_extrinsic_interactive.md
https://github.com/tier4/CalibrationTools/blob/tier4/universe/sensor/docs/images/camera-lidar/image_point_selection.png
https://github.com/tier4/CalibrationTools/blob/tier4/universe/sensor/docs/images/camera-lidar/calibration_1.png.

Since your initial tf looks good enough, the pointcloud points should appear close to where they should. If they do not appear at all, please verify that the python node is actually receiving them (ros2 topic info ... -v may be of help)

ajay1606, 2023-04-11:
@knzo25 First of all thank you so much for continuously assisting with your quick response. i greatly appreciate your patience and time. Also very sorry for bothering you so much with many silly questions sometimes. 

Am very pleased with the results and its very impressive compared to previous Autoware AI calibration and other online tools. Even relatively other calibration tools i have tested so far, this version appears to be so good. Though it is based on solve PnP (similar to Autoware AI), results really different than Autoware AI tools.

[image]

I will go through the code in detail, BTW If possible may i know, specific improvement added to this tool compared to AI tool previously released ?

knzo25, 2023-04-11:
:tada: :tada: :tada: 
Glad to know it is working for you !

Sadly (or happily), I only joined Tier IV by the time that we had already left the autoware AI era so I actually did not know that a similar tool existed until way after I wrote this tool. In the same way, this tool was not a part of our plans and I only dedicated it a few days since we needed to calibrate a sensor for some data that we had and I didn't want to do it tuning it manually.

Without knowing the changes, I can only guess, but:
 - Probably this tool allows you to get more precise image points due to the zom-in / zoom-out features.
 - OpenCV has changed since then, and maybe the algorithms themselves improved (not how there are two PNP implementations offered by the tool. The second one is quite new).
 - Maybe your data is just better this time. Remember that PNP assumes your intrinsics are correctly calibrated, so in the case that the intrinsics are wrong, a pose with low projection error does not exist.

If the projection error achieved by this tool proves to be insufficient for you, I recommend you to use our version that uses apriltags. I see you have a chessboard, but in our experiments we can achieve sub-pixel projection error with the apriltags.

 Remember to close this issue when you deem it resolved !

closed 2023-04-12 by ajay1606