tier4/CalibrationTools #100: Configuration for every Car and Sensor Kit model

danicannt, 2023-04-18:
Is there any guide on configuring the environment properly for a new use case? I have been trying to use your calibration tools using the ones that you have as an example trying to adapt them to our systems but I think that there is something I am missing out

knzo25, 2023-04-19:
@danicannt 
Sorry, even after your explanation I am unable to visualize the exact part you are having trouble with.
One thing we could do, as we have done before in previous issues, is that you fork the repo, prepare a branch stating what you want to do but does not work, and from there I can get a better picture of how to help. 

Additionally, from your 3 sensors, there are still various ways to create the tf tree so I would like to know how you envision it
e.g.,
lidar1 -> lidar2 -> camera

lidar1
 -> lidar2
 -> camera

camera
  -> lidar1
  -> lidar2

danicannt, 2023-04-21:
Sorry I've just realized that my last message was incomplete, I don't know if this would help you to give  some context to it:

> What we are trying to achieve is to configure a system that has two lidars (Ouster and Livox) and an RGB camera. As we are using the Autoware stack, what we wanted to do was to implement your calibration tools and be able to get the configuration files. I've tried to use the extrinsic_calibration_manager and for that, I configure my own folder of configurations following a model based on the aip_xx1 folder:
> 
> Inside the manual_sensor_kit_launch.xml that is loaded, I don't how to load/create these files for our system:
> 
```
  <!-- extrinsic_calibration_client -->
>   <arg name="src_yaml" default="$(find-pkg-share individual_params)/config/$(var vehicle_id)/$(var sensor_model)/sensor_kit_calibration.yaml"/>
>   <arg name="dst_yaml" default="$(env HOME)/sensor_kit_calibration.yaml"/>
> 
```

knzo25, 2023-04-24:
Sorry, I am still quite in the dark.

Are you asking about how to load the initial parameters?
If so, you can just modify the `src_yaml` path to suit your needs. In the above example we load it from the individual_params package since that is the way we organize the parameters, but you could do it in any way you consider it better.

Are you asking about how to format your yaml file?
That would depend on how you plan to structure your tf tree as I mentioned before.

ajay1606, 2023-05-03:
@danicannt Previously i also had similar issues in using this tool. But fortunately, @knzo25 continuous inputs helped me to figure out configuration required to run with custom sensor setup. Here are some highlights.

1. Make sure, static_tf published correctly, (You can write your own script to publish static tf params) #90 
2. Frames in launch file should correctly aligned with your TF tree.
I have tested all the calibration nodes, with @knzo25 help, and related discussion available here #90 #96

danicannt, 2023-05-10:
Sorry about the late reply, but I did not have time to be able to work in this project recently, I will try to work in it during this days.*

Btw, thanks for your answer @ajay1606 I will keep you both updated while I try

knzo25, 2023-06-12:
@danicannt 
As there has been no a activity in this issue, I will be closing it in about one week.

closed 2023-06-19 by knzo25