tier4/CalibrationTools #101: Camera LiDAR calibration error is large

CrisXin000, 2023-05-09:
Hello, thanks for posting such a great project. I tried to use `extrinsic_interactive_calibrator` for camera-lidar calibration, but the result looks bad, the closer to the edge of the image, the worse the calibration result is.
As shown in below:
[image]

During the calibration process, I placed three ARUCO markers in the camera field of view, and selected the corner points of the markers as the target points. 
Before calibration:
[image]

After calibration:
[image]

I have little experience with sensor calibration, can you give me some help?

CrisXin000, 2023-05-10:
Hi @knzo25 ,thanks for replying so quickly,

I understand the source of the calibration error from your answer. I can confirm the intrinsic calibration result is very good (the reprojection error is less than 0.2 pixels), so I think the camera-LiDAR calibration result maybe fall into a local optimum as you said (depending on the manually selected points), I will try placing more markers allows me to extract more image-lidar counterparts.

Also, I would include `tag-based calibrator` as a fallback if I'm not good at manually extracting it.

knzo25, 2023-06-12:
@CrisXin000 
As there has been no a activity in this issue, I will be closing it in about one week.

closed 2023-06-19 by knzo25