tier4/CalibrationTools #114: base_link to lidar not working

zillur-av, 2023-07-09:
I installed autoware and calibration packages from here. Everything seems to work fine. I am using a rosbag file that only contains `pointcloud2` messages. For now, my tf tree does not have anything, just an ouster lidar and base_link; I am not using `sensor_kit_base_link` for now. So, I modified `extrinsic_calibration_manager/launch/aip_x1/ground_plane.launch.xml` as follows:
```
<?xml version="1.0" encoding="UTF-8"?>
<launch>
  <arg name="vehicle_id" default="default"/>
  <let name="sensor_model" value="aip_x1"/>
  <let name="base_frame" value="base_link"/>
  <let name="parent_frame" value="base_link"/>
  <let name="rviz_profile" value="$(find-pkg-share extrinsic_ground_plane_calibrator)/rviz/velodyne_top.rviz"/>
  <arg name="rviz" default="true"/>

  <!-- extrinsic_calibration_client -->
  <arg name="src_yaml" default="$(find-pkg-share individual_params)/config/$(var vehicle_id)/$(var sensor_model)/sensor_kit_calibration.yaml"/>
  <arg name="dst_yaml" default="$(env HOME)/sensor_kit_calibration.yaml"/>

  <node pkg="extrinsic_calibration_client" exec="extrinsic_calibration_client" name="extrinsic_calibration_client" output="screen">
    <param name="src_path" value="$(var src_yaml)"/>
    <param name="dst_path" value="$(var dst_yaml)"/>
  </node>

  <!-- extrinsic_calibration_manager -->
  <node pkg="extrinsic_calibration_manager" exec="extrinsic_calibration_manager" name="extrinsic_calibration_manager" output="screen">
    <param name="parent_frame" value="$(var parent_frame)"/>
    <param name="child_frames" value="[os_sensor]"/>
  </node>

  <!-- velodyne_top_base_link: extrinsic_ground_plane_calibrator -->
  <group>
    <include file="$(find-pkg-share extrinsic_ground_plane_calibrator)/launch/calibrator.launch.xml">
      <arg name="ns" value="$(var parent_frame)/os_sensor"/>
      <arg name="base_frame" value="$(var base_frame)"/>
      <arg name="parent_frame" value="$(var parent_frame)"/>
      <arg name="child_frame" value="os_sensor"/>
      <arg name="pointcloud_topic" value="/ouster/points"/>
    </include>
  </group>

  <node pkg="rviz2" exec="rviz2" name="rviz2" output="screen" args="-d $(var rviz_profile)" if="$(var rviz)"/>
</launch>
```
And the calibration file is:
```
base_link:
  camera0/camera_link:
    x: 0.10731
    y: 0.56343
    z: -0.27697
    roll: -0.025
    pitch: 0.315
    yaw: 1.035
  os_sensor:
    x: 0.0
    y: 0.0
    z: 2.0
    roll: 0.0
    pitch: 0.0
    yaw: 0.0
```

[image]

I am publishing static transform from `base_link` to `os_sensor` like this in one terminal. It is just an initial guess.
```
trc@rahzar:~/autoware$ ros2 run tf2_ros static_transform_publisher 0 0 2 0 0 0 base_link os_sensor
[WARN] [1688935087.937881022] []: Old-style arguments are deprecated; see --help for new-style arguments
[INFO] [1688935087.962435324] [static_transform_publisher_6QasHP5sK0SW0rPO]: Spinning until stopped - publishing transform
translation: ('0.000000', '0.000000', '2.000000')
rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
from 'base_link' to 'os_sensor'
```
Another terminal, I am running the rosbag file. 

[image]

There is no error anywhere, still I do not think it is detecting `ground_plane` and just saying `waiting for the calibration to end`. How the calibration will end? Also, the output yaml file shows 0 for all 6DOF.

knzo25, 2023-07-09:
Hi, thanks for using our tool.

Apologies for making you tweak a lot of parts for it to launch correctly. We are aware that it is quite cumbersome and we are implementing a new design with more documentation, which is finally taking form, but it still would take a month or so at the least (company priorities).

Seeing your output, it seems that although the rough normal estimation through PCA was correctly detected (z component around 1 is good for more setups), it does not move much from there.

For example, if it does not find a plane, the following message should be displayed:
https://github.com/tier4/CalibrationTools/blob/81dd7fa86ebe7a05907593298c8dfd9b1e16483b/sensor/extrinsic_ground_plane_calibrator/src/extrinsic_ground_plane_calibrator.cpp#L286
And in case it is detected, it should display its parameters.

Are you by any chance compiling debug mode? Eigen and PCL are known to be extremely slow in that mode.
Another idea would be restricting the input space using a crop box  are shown here (this should make it faster)

https://github.com/tier4/CalibrationTools/blob/81dd7fa86ebe7a05907593298c8dfd9b1e16483b/sensor/extrinsic_ground_plane_calibrator/src/extrinsic_ground_plane_calibrator.cpp#L241

If these ideas do not work for you I can have a look at your data if your institution is ok with you sharing it.

zillur-av, 2023-07-09:
> Hi, thanks for using our tool.
>
> Apologies for making you tweak a lot of parts for it to launch correctly. We are aware that it is quite cumbersome and we are implementing a new design with more [...]

Thanks for your quick reply. I compiled it on release mode according to the instructions. Does the car have to standstill or I will drive it slowly while recording rosbag? The z axis from the ground plane should be around 2.5-3m. Even though it was showing 1m, the saved yaml shows all 0. Also, how do I stop the calibration process? Simply pressing ctrl+c? I can share data. I see the crop box filter is by default true. What can I do more?

knzo25, 2023-07-09:
The 1.0 showed in the console refers to the normal vector so 1.0 means that the lidar is mostly parallel with the floor (common in most sensor setup for the top lidar).

Although not really intended, if you press ctrl-c, the calibration manager dies without getting a result from the tool, and generates a file containing a null result (translation and orientation filled with zeros). It is misleading and will be replaced in the next version, so I will not change its current behavior until then.

This tool in particular is fully automatic, and with our sensors end in less than 10 seconds, so I did not consider making an external stopping interface.

The crop box is set to a +-50m, but in some indoor scenarios the floor (which is what we need to compute the base link's z-roll-pitch), is not seen after  10-20m. In those cases restricting the ranges makes it faster. This is not really necessary, but since it is not converging, it was just an idea to consider.

zillur-av, 2023-07-10:
I have used another rosbag file, this time found these: sometimes it detects plane and sometimes it doesn't. The process is just going on and on, it's not stopping itself and I had to press ctrl+c and below yaml file saved. Don't know about roll and pitch but z translation is wrong.
[image]
[image]

```
base_link:
  os_sensor:
    x: 0.000000
    y: 0.000000
    z: 0.001724
    roll: -0.018128
    pitch: -0.037037
    yaw: -0.000000
```

knzo25, 2023-07-11:
Hi, when the process doesn't stop is because the filter considers that the estimation has not converged. In the case that the detections are wrong, this is actually expected. Previously you mentioned that you can share the data. If that is still the case can you please post a link here?

zillur-av, 2023-07-11:
Sure, here you can find the rosbag lidar messages, YAML files : https://drive.google.com/drive/folders/1rw6yAA_XEaKni19OxmLHcpdxUR7Bk7ld?usp=sharing

knzo25, 2023-07-11:
Thanks for sharing, I will check it tomorrow and come back to you :)

knzo25, 2023-07-12:
Hi, I had a look at your data.

It seems ouster sensors have a little bit more noise than the sensors I had been using, which ended up making RANSAC fail.
In the file `sensor/extrinsic_ground_plane_calibrator/launch/calibrator.launch.xml` I just changed the following parameter

```
<param name="max_inlier_distance" value="0.05"/>
```

(branch `feature/ground_plane_redesign`)

[image]

knzo25, 2023-07-12:
@zillur-av 

Hi, I had a look at your data and it seems the ouster has a little bit more noise than the lidars I am currently using.
As such, in the file `sensor/extrinsic_ground_plane_calibrator/launch/calibrator.launch.xml` I just changed the following parameter:

```
<param name="max_inlier_distance" value="0.05"/>
```

(branch: `feature/ground_plane_redesign`)

[image]

zillur-av, 2023-07-12:
Appreciate your time and efforts. Could not find a better flat surface nearby. So, after changing that parameter,  is it working properly? I will check tomorrow morning. Hopefully,  my config files and everything else are ok.

knzo25, 2023-07-12:
Yes, it works as shown in the previous image.
The  white points are the detected ground, which is used to fit a plane, which in turn is used to project the lidar, finding the "base_link" (bar the x, y, and yaw components).

This tool was made for a specific product that has a controlled  environment, hence the parameters not being suited for just any environment. As such, you may have to play with the parameters in the before mentioned launcher.

Please let us know how if it works for you and if you have any other questions (and in the case you have no other concerns, please remember to close the issue)

zillur-av, 2023-07-12:
Thanks you so much for your help. It's working fine now and the z is almost accurate. Not sure about roll and pitch though as I could not measure them manually.

closed 2023-07-12 by zillur-av