tier4/CalibrationTools #116: fix(calibration_tools.repos): github permission denied problem

yvzksgl, 2023-07-27:
Hello, thanks to everyone this amazing tool. I have a suggestion for calibration_tools.repos file.

When the command ```vcs import src < calibration_tools.repos``` i got github permission denied error.

Here is my suggestion for calibration_tools.repos:

```
repositories:
  autoware/calibration_tools:
    type: git
    url: https://github.com/tier4/CalibrationTools
    version: tier4/universe
  vendor/lidartag:
    type: git
    url: https://github.com/tier4/LiDARTag.git
    version: humble
  vendor/lidartag_msgs:
    type: git
    url: https://github.com/tier4/LiDARTag_msgs
    version: tier4/universe
  vendor/apriltag_msgs:
    type: git
    url: https://github.com/christianrauch/apriltag_msgs
    version: 2.0.0
  vendor/apriltag_ros:
    type: git
    url: https://github.com/christianrauch/apriltag_ros
    version: e814e9e5d5f1bfb60a4aa685d30977c632bbc540
  vendor/ros2_numpy:
    type: git
    url: https://github.com/Box-Robotics/ros2_numpy
    version: humble
    vendor/image_pipeline:
    type: git
    url: https://github.com/tier4/image_pipeline.git
    version: 47964112293eb19f9f57254b2e6b68706954cc63
```

knzo25, 2023-07-27:
Hi, thanks for your contribution !
At some point in the past we had most of the repositories with `https` instead of `git` and got told by someone in our organization that get should use `git` as much as possible. Let me ask again and come back to you :bow:

aohsato, 2023-07-27:
Thanks for the great feedback. Since these repositories are public, it's better to use `https` which does not require ssh key registration.

yvzksgl, 2023-07-27:
Thanks for your quick responses. One extra comment from me, I thought github was ending support for ssh keys. Looks like I'm misremembering, I'll look into this as well. Thank you all.

yvzksgl, 2023-08-01:
Sure, i was waiting for merge. Thanks for the information.

closed 2023-08-01 by yvzksgl