tier4/CalibrationTools #118: Estimation of EKF Localizer Parameters

yvzksgl, 2023-07-28:
Hello,

I have a question about how to estimate ```ekf_localizer``` parameters with using ```deviation_estimator``` tool. Readme says: 
```The node also estimates the standard deviation of velocity and yaw rate. This can be used as a parameter in ```ekf_localizer.```
But the node's outputs are imu_corrector.param.yaml and vehicle_velocity_converter.param.yaml files' parameters. Can I transform 
```
velocity_stddev_xx
angular_velocity_stddev_xx
angular_velocity_stddev_yy
angular_velocity_stddev_zz
```
to
```
proc_stddev_yaw_c
```
In short i want to use deviation_estimator tool to optimize ekf_localizer parameters, but i cannot find a way to do that.

knzo25, 2023-07-28:
@kminoda :pray:

kminoda, 2023-07-31:
`proc_stddev_yaw_c` is a process noise and not observation noise. It is irrelevant to the sensors, but to the kinematics model. 
The estimated covariance for IMU and velocity should be reflected in `imu_corrector` and `vehicle_velocity_converter`, respectively. Those would be eventually merged into `/localization/twist_estimator/twist_with_covariance` with the covariance reflected.

yvzksgl, 2023-07-31:
İ understand thank yo for your response. Have a nice week.