tier4/CalibrationTools #136: <Build Error> tier4_autoware_utils/tier4_autoware_utils.hpp: No such file or directory,

knorrrr, 2023-10-18:
When I tried to build extrinsic_reflector_based_calibrator, I encounterd this error message.

I removed this and I succeeded to compile. I don't know this process is correct.
```
tier4_autoware_utils/tier4_autoware_utils.hpp: 
16 | #include <tier4_autoware_utils/tier4_autoware_utils.hpp>
```

Error Message:
```
 colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release --packages-select extrinsic_reflector_based_calibrator
Starting >>> extrinsic_reflector_based_calibrator
--- stderr: extrinsic_reflector_based_calibrator
/autoware/src/autoware/calibration_tools/sensor/extrinsic_reflector_based_calibrator/src/extrinsic_reflector_based_calibrator.cpp:16:10: fatal error: tier4_autoware_utils/tier4_autoware_utils.hpp: No such file or directory
   16 | #include <tier4_autoware_utils/tier4_autoware_utils.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/extrinsic_reflector_based_calibrator.dir/build.make:76: CMakeFiles/extrinsic_reflector_based_calibrator.dir/src/extrinsic_reflector_based_calibrator.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:192: CMakeFiles/extrinsic_reflector_based_calibrator.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< extrinsic_reflector_based_calibrator [7.56s, exited with code 2]
```

knzo25, 2023-10-18:
Hi, thanks for the report.
The fix was proposed in https://github.com/tier4/CalibrationTools/pull/135

knorrrr, 2023-10-18:
I wish this pull request will be merged.
Great Thanks!

knzo25, 2023-10-19:
Fixed in https://github.com/tier4/CalibrationTools/pull/135

closed 2023-10-25 by knorrrr