tier4/CalibrationTools #159: Interactive calibration tool does not work. White screen

javierAraluce, 2024-04-04:
Hi there,
I am trying to use your framework to perform a camera-lidar calibration, but I am struggling with some issues.

The interactive calibration tool stays white as you can see in the picture below. The callback does something strange because I code a print() for debugging and it prints nothing until I kill the process. At that moment it prints all the messages.

[image]

These are the commands that I execute:
`ros2 launch extrinsic_calibration_manager calibration.launch.xml mode:=interactive sensor_model:=robobus vehicle_model:=tutorial_vehicle vehicle_id:=robobus camera_name:=camera_front`

`ros2 bag play rosbag2_2024_04_04-06_25_04/ --clock -l -r 0.2 --remap /tf:=/null/tf /tf_static:=/null/tf_static /robobus/camera_front/image_raw:=/robobus/sensor_kit_base_link/camera_front/image_raw /robobus/camera_front/camera_info:=/robobus/sensor_kit_base_link/camera_front/camera_info /robobus/ouster_points_1:=/robobus/sensor_kit_base_link/ouster_points_1`

I did a remapping to see if the namespaces were incorrect, but nothing changed.
The output of the rqt_graph:
[image]

Thanks in advance

knzo25, 2024-04-09:
Thank you for using our stack

Regarding your question, although you provided relevant information, in this case it is not sufficient to give you a proper answer.
I would recommend:
 - Checking that all the subscribers from the node have data coming in
 - The related tfs exist
 - That the stamps between the lidar and camera are synchronized. This last point is usually the reason why no image appears. I added as a requirement for it to be sync (so we can not attempt to do calibration under wrong data). In the PR to be merged some more information should be displayed, like the delay between messages (https://github.com/tier4/CalibrationTools/pull/148).

Alternatively, if your institution allows it, we can also try to reproduce your error with your data.

javierAraluce, 2024-04-17:
Thank you for your reply. I will go through the points in detail:

- I have checked that all topics are published because I can visualise them on Rviz.
- I have a question about the tfs. According to the tutorial, the original tf from the rosbag should be remapped, but without them I cannot see anything on Rviz. What is the correct procedure? 
- Each sensor was recorded with the same computer, I did not check if they were synchronised, but they should be.

I can send you a rosbag, how do I transfer it to you and how long does it have to be?

Thanks again for your amazing work.

knzo25, 2024-04-17:
Thanks for your answer.

Regarding the remapping, in autoware, our rosbags contain /tf (localization) and /tf_static (extrinsics).
Since autoware may use different parameters, different from the ones present in the rosbag, we normally recommend the remap. However, that is up to the user (we will need to clarify this in the documentation)

The rosbag should be just a dozen of seconds, containing camera info, images, and pointclouds in PointCloud2 format. 
I am not sure if we can share our emails in github, but in the past users have provided rosbags via public means (university servers, google drive, etc) and then deleted the links as soon as we notified them that we got the data

javierAraluce, 2024-04-23:
Hello again, 

I have recorded a ros2 bag with the expected topics. Thanks for the tf clarification.

The file will be deleted in 7 days.
https://we.tl/t-OWfYtwDm29

javierAraluce, 2024-04-23:
Hello again,

I have some news, after recording some rosbags to send to you, I discovered the bug. It was that the point cloud and camera were not synchronised because the Ouster driver by default uses the time from boot instead of the ROS time. Rviz shows the topics fine, but the calibration does not. To fix this, check the timestamps of both topics and where they are taken from.

Thanks for your feedback

knzo25, 2024-04-23:
@javierAraluce 
If you are satisfied with your solution, could you close the issue?

javierAraluce, 2024-04-24:
Thanks for your help.

closed 2024-04-24 by javierAraluce