tier4/CalibrationTools #163: Saving of .yaml not possible

Sophie2216, 2024-04-19:
Hello,
After the calibration of the intrinsic parameters by the intrinsic_camera_calibrator is done and the Calibration Status turned back to idle, I wanted to save the calibrated intrinsic parameters by pressing the button 'Save'. Unfortunately, it just shows that a window pops up but as you can see in the picture no window is there.

[image]

How can I save the calibrated Intrinsics?

Thanks for your support in advance!

knzo25, 2024-04-19:
@Sophie2216 
Hi,
I have observed that behavior a few times last year, but I have not been able to reproduce it since.

I would recommend you to disable the drawings after calibrating, they events may be blocking the other UI.
If not, please close the rosbag or the camera driver, that worked to me last year.

Please let us know if it works for you or if you have clear ways to reproduce so we can try to fix it !

knzo25, 2024-05-24:
Since a month has passed without any updates I will be closing this issue. Please re-open as needed

closed 2024-05-24 by knzo25