tier4/CalibrationTools #169: extrinsic_calibration_manager can‘t find!

0116lhc, 2024-04-24:
The package extrinsic_calibration_manager cannot be found in the sensor folder. I've pulled the code several times but it's still missing. Could you please confirm the location of this package?

yoepav, 2024-05-24:
Hello Kenzo,

Can I follow the current documentation for manual calibration and change the steps regarding the extrinsic_calibration_manager for the sensor_calibration_manager?

knzo25, 2024-05-24:
Hi,
Thank you for using our tools. Sadly, I am not the maintainer of the manual calibration tool, so its use under the new scheme will need to wait until the maintainer ports the tool to the new API (or, if you prefer, you could use a previous version of the repository that is compatible with the manual tool)

yoepav, 2024-05-28:
Hi Kenzo,

I tried using the main branch instead of the tier4/universe branch but sadly that doesn't work. So I guess I'll have to wait for the port. Thank you very much anyway.

dtonda8, 2024-07-07:
Hi @yoepav & @0116lhc,

The latest Autoware.Universe repo had some package renaming, which causes the rosdep dependency issues you've listed.

I've [forked the CalibrationTools](https://github.com/dtonda8/CalibrationTools) repo with the correct dependency package naming and no longer experience the dependency issues. My fork also reverted to the older version CalibrationTools (i.e. with `extrinsic_calibration_manager` and `aip_x1` directory). I hope this can be useful untill the re-written calibration docs are ready.

TimDing-aoi, 2024-08-15:
> Hi @yoepav & @0116lhc,
>
> The latest Autoware.Universe repo had some package renaming, which causes the rosdep dependency issues you've listed.
>
> I've [forked the [...]

Hello dtonda, thanks for the upload, I have tried your version of the calibration tools but fails to colcon build. 
[image]
Do you know what might be the reason for this?

dtonda8, 2024-08-15:
Hi @TimDing-aoi, I've created a new issue for [this in my fork](https://github.com/dtonda8/CalibrationTools/issues/1) (it's most due to Calibration repos referring to incorrect package naming). If you find out where exactly the issue lies and resolve it, you can create a pr. I will also look into this when I am free.

knzo25, 2024-08-19:
@dtonda8 
I am pretty sure the code in this PR works
https://github.com/tier4/CalibrationTools/pull/164

dtonda8, 2024-08-19:
Yes, that pr seems to work. The only issue is that you'll need to switch to a slightly older version of autoware.universe, otherwise the output of `rosdep install -y --from-paths src --ignore-src --rosdistro $ROS_DISTRO` is this:
[image]

Autoware.universe recently renamed `kalman_filter` to `autoware_kalman_filter` in this [commit](https://github.com/autowarefoundation/autoware.universe/commit/2f392a849c7f93d2bb0b1305dec1fa2d9a5bd099)
So we should `git checkout` to a commit before that ([fd09a5b](https://github.com/autowarefoundation/autoware.universe/commit/fd09a5bb1daea52afbe2bea11402f87e3d7eb339))
[image]

knzo25, 2024-08-20:
@dtonda8 
I just updated the PR branch to be compatible with the latest universe (https://github.com/tier4/CalibrationTools/pull/164/commits/1dfc8d72728fb32867fb88af8ee954d6aae62c34)

dtonda8, 2024-08-20:
@knzo25 Legend! Thanks for doing that. To everyone else looking for a CalibrationTools version compatible with the latest Autoware and matches the current `extrinsic_calibration_manager` docs, please use knzo25's pr branch.

knzo25, 2024-09-24:
These issues should disappear in the main branch now
https://github.com/tier4/CalibrationTools/pull/164

closed 2024-09-24 by knzo25