tier4/CalibrationTools #175: rosdep cannot locate some dependencies

zichongy, 2024-06-29:
Hi,

I was trying to install these calibration tools after installing Autoware.Universe following the documentation. However, I encountered rosdep errors shown below.

```
$ rosdep install -y --from-paths `colcon list --packages-up-to sensor_calibration_tools -p` --ignore-src
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
ground_plane_calibrator: Cannot locate rosdep definition for [tier4_autoware_utils]
tag_based_sfm_calibrator: Cannot locate rosdep definition for [tier4_autoware_utils]
mapping_based_calibrator: Cannot locate rosdep definition for [tier4_autoware_utils]
lidar_to_lidar_2d_calibrator: Cannot locate rosdep definition for [tier4_autoware_utils]
tag_based_pnp_calibrator: Cannot locate rosdep definition for [tier4_autoware_utils]
sensor_calibration_tools: Cannot locate rosdep definition for [extrinsic_map_based_calibrator]
tier4_ground_plane_utils: Cannot locate rosdep definition for [tier4_autoware_utils]
marker_radar_lidar_calibrator: Cannot locate rosdep definition for [tier4_autoware_utils]
```

I tried to update rosdep, but it seems it's not working. After investigation, I found out that tier4_autoware_utils was removed from the latest Autoware documentation. Will there be a breach with updated dependencies or how can I manually edit the dependencies to solve the issue?

knzo25, 2024-06-29:
@zichongy 
We are currently writing a new documentation for the tools, which has taken more than expected.
While doing so, there have been breaking changes outside this repository, and while not the cleanest way of doing things, we have been adding the fixes there (as they affected the documentation)

In summary, please use the branch new_api_documentation branch until the telated PR gets merged #164 

If this works for you, please consider closing the issue later 🙏

zichongy, 2024-06-30:
Hi @knzo25,

Thanks for the reply. I tried the following steps again. But it still didn't work.

1. Deleted all previously imported folders in `/src`
2. Downloaded the branch you described by `wget https://raw.githubusercontent.com/tier4/CalibrationTools/feature/new_api_documentation/calibration_tools_autoware.repos`
3. `vcs import src < calibration_tools_autoware.repos`
4. Perform `rosdep install -y --from-paths src --ignore-src --rosdistro $ROS_DISTRO` again. But it  still shows the same error message.
```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
lidar_to_lidar_2d_calibrator: Cannot locate rosdep definition for [tier4_autoware_utils]
tag_based_sfm_calibrator: Cannot locate rosdep definition for [tier4_autoware_utils]
tag_based_pnp_calibrator: Cannot locate rosdep definition for [tier4_autoware_utils]
ground_plane_calibrator: Cannot locate rosdep definition for [tier4_autoware_utils]
sensor_calibration_tools: Cannot locate rosdep definition for [extrinsic_map_based_calibrator]
tier4_ground_plane_utils: Cannot locate rosdep definition for [tier4_autoware_utils]
mapping_based_calibrator: Cannot locate rosdep definition for [tier4_autoware_utils]
tunable_static_tf_broadcaster: Cannot locate rosdep definition for [tier4_autoware_utils]
marker_radar_lidar_calibrator: Cannot locate rosdep definition for [tier4_autoware_utils]
```

How can I ensure that all previously imported files were deleted? Or am I downloading the correct `.repos` file?

knzo25, 2024-06-30:
Hi, the problem is that 
https://raw.githubusercontent.com/tier4/CalibrationTools/feature/new_api_documentation/calibration_tools_autoware.repos
is meant to work once merged. As you can see from the contents, it is still trying to download the main branch.
In this case (and in the meantime), please checkout the PR branch between steps 2 and 3

zichongy, 2024-07-01:
Thanks for the instruction. Closing the issue.

closed 2024-07-01 by zichongy

dtonda8, 2024-07-07:
The latest Autoware.Universe repo had some package renaming, which causes the rosdep dependency issues you've listed.

I've [forked the CalibrationTools](https://github.com/dtonda8/CalibrationTools) repo with the correct dependency package naming and no longer experience the dependency issues. My fork also reverted to the older version CalibrationTools (i.e. with `extrinsic_calibration_manager` and `aip_x1` directory). I hope this can be useful untill the re-written calibration docs are ready.