tier4/CalibrationTools #177: Radar - LiDAR Calibration docs

AhmetMete, 2024-07-04:
Hi @knzo25,

I want to try Radar - LiDAR calibration. But i could not find Radar-LiDAR calibration readme in the docs. 

Can you explain how can i try this calibration method?

Can you give some video, report  or like some stuff ?

Thank you for your help.

knzo25, 2024-07-04:
@AhmetMete 
The calibration is still being written.
That being said, you can access the current one in the respective PR (there is some data you can use).
Out of curiosity, what radar/lidar are you using?

https://github.com/tier4/CalibrationTools/tree/feature/new_api_documentation/calibrators/marker_radar_lidar_calibrator
https://github.com/tier4/CalibrationTools/blob/feature/new_api_documentation/calibrators/docs/tutorials/marker_radar_lidar_calibrator.md

AhmetMete, 2024-07-04:
Hello again, 

Thank you for your reply. 

I am using Ouster LiDAR and SmartMicro radar. 

Lastly I have a question. Do you have an estimated time for the end of the calibration package? 

Best Regards.

knzo25, 2024-07-04:
What do you mean by `end`?
If it is merge/release, it was planned for a few months back, but the documentation writing has not moved as planned. As such, I can not give you a concrete date. Hopefully before august.

By the way, since we only have continental radars here, we can only guarantee compatibility with that one. If you can provide us data (compatible with our process), we may be able to add support

AhmetMete, 2024-07-05:
I meant merge date. 

Thank you for all your help. 

We can also contact again in the future about data sharing 

Best Regards.

knzo25, 2024-07-05:
@AhmetMete 
In that case, can you close the issue for now ?
(you can always create another one later)

AhmetMete, 2024-07-05:
Yes, sure. 

Thank you for your help.

AhmetMete, 2024-08-19:
Hi again @knzo25,

I want to ask, is **radar calibration** package development finished?

I can't find new commits related **radar calibration** in main branch.

Best Regards.

knzo25, 2024-08-19:
The @AhmetMete 
The package was made for the ARS408. It will probably not fit other sensors our of the box (it was made in about a week to fit our needs). As such, I leave up to you whether it can be considered as finished. We have another PR to add support to other sensors, but then again, we do not provide any guarantees for any other radars

Our main branch is not `main` to probably that is why you can not find it.
[image]
https://github.com/tier4/CalibrationTools/pull/108

AhmetMete, 2024-08-22:
Thank you for your reply.

closed 2024-08-22 by AhmetMete