tier4/CalibrationTools #181: How can I receive extrinsic_calibration service?

enddl3224, 2024-07-23:
Hi, I'm trying to use camera-lidar calibration tool using `tag_based_pnp_calibrator.launch.xml`.
and I want to check the values(initial_reprojection_error, current_reprojection_error, filtered_reprojection_error).

If I receive `extrinsic_calibration` service,
then `request_received_ = true;` 
I think I can also get the above values as well.

How can I receive `extrinsic_calibration` service?

I'm attaching some pictures.

- Code
[image]
[image]

- Image view
[image]

- TF tree
[image]

- node, topic info
[image]
[image]
[image]
[image]

Thank you.

knzo25, 2024-07-25:
@enddl3224 
Although you provided a lot of data I don't fully understand.
Are you using the calibrator manager? If not, you can always send the service manually

enddl3224, 2024-07-25:
Thank you for comment!
I'm using the calibrator manager(sensor_calibration_manager).
But could you tell me how can I send the service manually?
I guess I didn't send the service.

enddl3224, 2024-07-29:
Thank you so much! I'll try it.

knzo25, 2024-08-12:
@enddl3224 
If there are no updated on this, I will be closing this in about one week