tier4/CalibrationTools #183: Issue with rosdep installation of Autoware calibration tools

youweik2, 2024-07-29:
I have successfully installed Autoware on my system running Ubuntu 22.04 and ROS 2 Humble. However, I'm facing issues when trying to install the calibration tools. The command I ran is:
rosdep install -y --from-paths src --ignore-src --rosdistro $ROS_DISTRO

I received the following error message:

ERROR: the following packages/stacks could not have their rosdep keys resolved to system dependencies:
tunable_static_tf_broadcaster: Cannot locate rosdep definition for [tier4_autoware_utils]
lidar_to_lidar_2d_calibrator: Cannot locate rosdep definition for [tier4_autoware_utils]
ground_plane_calibrator: Cannot locate rosdep definition for [tier4_autoware_utils]
sensor_calibration_tools: Cannot locate rosdep definition for [extrinsic_map_based_calibrator]
mapping_based_calibrator: Cannot locate rosdep definition for [tier4_autoware_utils]
tag_based_pnp_calibrator: Cannot locate rosdep definition for [tier4_autoware_utils]
marker_radar_lidar_calibrator: Cannot locate rosdep definition for [tier4_autoware_utils]
tier4_ground_plane_utils: Cannot locate rosdep definition for [tier4_autoware_utils]
tag_based_sfm_calibrator: Cannot locate rosdep definition for [tier4_autoware_utils]

It seems that multiple packages are missing their rosdep definitions, specifically for tier4_autoware_utils and extrinsic_map_based_calibrator.

Could anyone provide guidance on how to resolve these missing dependencies? I'm quite new in this field, so I don't know how to solve it.

Thanks for your help!

knzo25, 2024-07-30:
@youweik2 
Thanks for raising the issue. As of now, the `tier4/universe` is broken and fixes were added in the documentation branch (See #164 ). Please use that in the meantime :bow:

youweik2, 2024-07-30:
> @youweik2 Thanks for raising the issue. As of now, the `tier4/universe` is broken and fixes were added in the documentation branch (See #164 ). Please use that in the meantime 🙇

Got it!

youweik2, 2024-07-30:
> @youweik2 Thanks for raising the issue. As of now, the `tier4/universe` is broken and fixes were added in the documentation branch (See #164 ). Please use that in the meantime 🙇

Sorry to bother you again. I tried again today with the new branch, but there remains the same error of rosdep. Both autoware or non-autoware interfaces didn't work on my system.

knzo25, 2024-08-01:
@youweik2 
I just tried and it compiled.
However, you need to remember that the PR is meant to be merged so there are two things you need to do:
 - Use https://raw.githubusercontent.com/tier4/CalibrationTools/feature/new_api_documentation/calibration_tools_standalone.repos instead of the one from the `tier4/universe` branch.
 -  Checkout the correct branch (`feature/new_api_documentation`) after vcs

youweik2, 2024-08-01:
> Sorry to bother you again. I tried again today with the new branch, but there remains the same error of rosdep. Both autoware or non-autoware interfaces didn't work on my system.

Sorry for any confusion.I have used wget to download the calibration_tools_autoware.repos file from the branch feature/new_api_documentation and followed the provided instructions. However, I also tried downloading the calibration_tools_standalone.repos file into a new directory, but encountered the same error.

Here’s the error message I received when running rosdep (still same):

auto2204@auto:~/calibration$ rosdep install -y --from-paths `colcon list --packages-up-to sensor_calibration_tools -p` --ignore-src
ERROR: the following packages/stacks could not have their rosdep keys resolved to system dependencies:
ground_plane_calibrator: Cannot locate rosdep definition for [tier4_autoware_utils]
sensor_calibration_tools: Cannot locate rosdep definition for [extrinsic_map_based_calibrator]
marker_radar_lidar_calibrator: Cannot locate rosdep definition for [tier4_autoware_utils]
tier4_ground_plane_utils: Cannot locate rosdep definition for [tier4_autoware_utils]
mapping_based_calibrator: Cannot locate rosdep definition for [tier4_autoware_utils]
lidar_to_lidar_2d_calibrator: Cannot locate rosdep definition for [tier4_autoware_utils]
tag_based_sfm_calibrator: Cannot locate rosdep definition for [tier4_autoware_utils]
tag_based_pnp_calibrator: Cannot locate rosdep definition for [tier4_autoware_utils]

I don't know why I couldn't make it. Thank you so much for your reply!

TZECHIN6, 2024-08-06:
@youweik2 after you have wget the the calibration.repos, by default no matter the standalone or autoware version, both of them point to tier4/universe. One of the easy way is to cd into that repo and git switch to `feature/new_api_documentation`.

After that, rerun the cmd of `rosdep install ...`, there should be several packages you have to install locally. Once everything is installed, you should see All dependencies are installed. Now you can `colcon build` the workspace.

youweik2, 2024-08-06:
> @youweik2 after you have wget the the calibration.repos, by default no matter the standalone or autoware version, both of them point to tier4/universe. One of the easy way is to cd into that repo [...]

Thank you so much! However, after running git switch feature/new_api_documentation and git checkout, the cmd of rosdep install return a new error as follow:
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
sensor_calibration_manager: Cannot locate rosdep definition for [image_transport_decompressor]

Thank you again for your reply!

TZECHIN6, 2024-08-07:
Are you running it standalone or using with autoware? `image_transport_decompressor` is installed along with autoware.universe. If yes, which version (commit hash) of universe are you installed?

youweik2, 2024-08-07:
> Are you running it standalone or using with autoware? `image_transport_decompressor` is installed along with autoware.universe. If yes, which version (commit hash) of universe are you installed?

I run it with Autoware. My autoware.universe version is 1640c06dacf9958fd0da03d2b23a9abc887d3cf2 and the branch is main. I just followed the autoware installation guidance to install it.

ajay1606, 2024-08-08:
@knzo25 @TZECHIN6 I also tried build the repo as you mentioned for [standalone_repo], (https://raw.githubusercontent.com/tier4/CalibrationTools/feature/new_api_documentation/calibration_tools_standalone.repos) but ended with similar error.  

```
rosdep install -y --from-paths `colcon list --packages-up-to sensor_calibration_tools -p` --ignore-src
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
sensor_calibration_manager: Cannot locate rosdep definition for [image_transport_decompressor]
```
Any help much appreciated.

TZECHIN6, 2024-08-12:
@youweik2 @ajay1606 https://github.com/autowarefoundation/autoware.universe/pull/8197
I believe they update the prefix of this package, for your own project, would you please check the `image_transport_decompressor` under autoware.universe/sensing/...

You might manually update the package name according to what its shown under your project.

knzo25, 2024-08-12:
@ajay1606 @youweik2 @TZECHIN6 

Sorry, I have been out for a while, which ended up with delays in my replies here.
I just updated the branch with the `autoware_` prefix. When the associated PR has been merged, we plan to start using versioning between universe and this repository since it is prone to keep breaking.

Regarding the numpy issue, please check the output of this:

```bash
username@hostname:~/workspace/calibration_tools_ws_issue$ which python
/usr/bin/python
username@hostname:~/workspace/calibration_tools_ws_issue$ python
Python 3.10.12 (main, Jul 29 2024, 16:56:48) [GCC 11.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import numpy as np
>>> np.__version__
'1.21.5'
```

If you are using a different kind of environment, you may want to use the docker version

ajay1606, 2024-08-12:
@knzo25 Thanks for your kind response. But i didnt understand your modification. Problem remain same 

[image]

I am currently trying to build and run this repo standalone, though packages built successfully.

Command `ros2 run sensor_calibration_manager sensor_calibration_manager` failed to launch calibration tool.

Would you please help us to solve this.

Currently i am able to checkout  `feature/new_api_documentation` branch, build successful.

knzo25, 2024-08-12:
@ajay1606 
Like I mentioned in the previous post, your problem is related to the numpy version of your environment

closed 2024-08-13 by youweik2

ajay1606, 2024-08-13:
> @ajay1606 Like I mentioned in the previous post, your problem is related to the numpy version of your environment

@knzo25 @TZECHIN6 Tool working after changing numpy version.

knzo25, 2024-09-03:
I closed this issue since it was due to the version of the dependencies and it did not appear on my environment, but if you do not want/can not downgrade numpy, the following also helps:

```bash
kenzolobos@PC:~/workspace/calibration_ws$ pip list | grep numpy
numpy                                1.26.4
ros2_numpy                           2.0.11
kenzolobos@PC:~/workspace/calibration_ws$ 
kenzolobos@PC:~/workspace/calibration_ws$ ros2 run sensor_calibration_manager sensor_calibration_manager 
Traceback (most recent call last):
  File "/home/kenzolobos/workspace/calibration_ws/install/sensor_calibration_manager/lib/sensor_calibration_manager/sensor_calibration_manager", line 33, in <module>
    sys.exit(load_entry_point('sensor-calibration-manager', 'console_scripts', 'sensor_calibration_manager')())
  File "/home/kenzolobos/workspace/calibration_ws/install/sensor_calibration_manager/lib/sensor_calibration_manager/sensor_calibration_manager", line 25, in importlib_load_entry_point
    return next(matches).load()
  File "/usr/lib/python3.10/importlib/metadata/__init__.py", line 171, in load
    module = import_module(match.group('module'))
  File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/kenzolobos/workspace/calibration_ws/build/sensor_calibration_manager/sensor_calibration_manager/sensor_calibration_manager.py", line 49, in <module>
    from sensor_calibration_manager.calibrator_base import CalibratorState
  File "/usr/lib/python3/dist-packages/shiboken2/files.dir/shibokensupport/__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "/home/kenzolobos/workspace/calibration_ws/build/sensor_calibration_manager/sensor_calibration_manager/calibrator_base.py", line 29, in <module>
    import transforms3d
  File "/usr/lib/python3/dist-packages/shiboken2/files.dir/shibokensupport/__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "/usr/lib/python3/dist-packages/transforms3d/__init__.py", line 10, in <module>
    from . import quaternions
  File "/usr/lib/python3/dist-packages/shiboken2/files.dir/shibokensupport/__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "/usr/lib/python3/dist-packages/transforms3d/quaternions.py", line 26, in <module>
    _MAX_FLOAT = np.maximum_sctype(np.float)
  File "/home/kenzolobos/.local/lib/python3.10/site-packages/numpy/__init__.py", line 324, in __getattr__
    raise AttributeError(__former_attrs__[attr])
AttributeError: module 'numpy' has no attribute 'float'.
`np.float` was a deprecated alias for the builtin `float`. To avoid this error in existing code, use `float` by itself. Doing this will not modify any behavior and is safe. If you specifically wanted the numpy scalar type, use `np.float64` here.
The aliases was originally deprecated in NumPy 1.20; for more details and guidance see the original release note at:
    https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations. Did you mean: 'cfloat'?
[ros2run]: Process exited with failure 1
kenzolobos@PC:~/workspace/calibration_ws$ pip install transforms3d
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: transforms3d in /usr/lib/python3/dist-packages (0.3.1)
kenzolobos@PC:~/workspace/calibration_ws$ pip install transforms3d --upgrade
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: transforms3d in /usr/lib/python3/dist-packages (0.3.1)
Collecting transforms3d
  Downloading transforms3d-0.4.2-py3-none-any.whl (1.4 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.4/1.4 MB 47.4 MB/s eta 0:00:00
Requirement already satisfied: numpy>=1.15 in /home/kenzolobos/.local/lib/python3.10/site-packages (from transforms3d) (1.26.4)
Installing collected packages: transforms3d
Successfully installed transforms3d-0.4.2
kenzolobos@PC:~/workspace/calibration_ws$ ros2 run sensor_calibration_manager sensor_calibration_manager 
INFO:root:Closing calibrator selector view
INFO:root:Received sigint. Quitting...
INFO:root:No launch process to terminate
```