tier4/CalibrationTools #185: standalone repo with rosbag simulation

ajay1606, 2024-08-20:
@knzo25 Thanks helping us to solve previous issue. 

We able to build and run as you suggested, but while trying as mentioned in the document (with sample ros bag) unable to see the tf tree as it mentioned in the document.

[image]

We are currently testing with `feature/new_api_documentation` repo. Should we need to publish senor TF seperately ?

Appreciate any hlep !

Platform: 
`ROS2 Humble`,

knzo25, 2024-08-23:
Without more information I would think the tfs are not being published.
Whether they are / should be embedded is the rosbag depends on your pipeline

ajay1606, 2024-08-27:
@knzo25 Thanks for your response, actually we are trying to run with sample ros bag mentioned in the document.

knzo25, 2024-08-27:
That is weird as we have tried that before. Can you give the output of:
 - ros2 bag info
 - ros2 run tf2_tools view_frames (screenshot of the resulting pdf file)
 - confirmation that you are using the x2 project
 - full console logs
 ?

ajay1606, 2024-08-28:
@knzo25 Sorry for the late response as I was out of lab for few days. 
I have carefully followed as it [here](https://github.com/tier4/CalibrationTools/blob/feature/new_api_documentation/calibrators/docs/tutorials/tag_based_pnp_calibrator.md)

Tf tree lloks like this:

[image]

[image]

[image]

Still unable to see the tf tree in GUI.
[image]

Also, able to see data publishing in the topic using `ros2 topic echo ` but unable to visualize as getting warning like below;
[image]

Any suggestions please ! Greatly appreciated.

knzo25, 2024-08-30:
@ajay1606 
I just tried again in a different setup with a clean installation and work as expected.
It concerns me that the "Not ready" label should not appear if the calibration services or the tfs are available.

I used:
`ros2 run sensor_calibration_manager sensor_calibration_manager`
`ros2 bag play camera_lidar.db3 --clock -r 0.2`

For reference, this is the output on my machine:
```text
INFO:root:on_selected_calibrator: project_name=x2 calibrator_name=tag_based_pnp_calibrator
INFO:root:Reading xml from: /home/kenzolobos/workspace/calibration_ws/install/sensor_calibration_manager/share/sensor_calibration_manager/launch/x2/tag_based_pnp_calibrator.launch.xml
INFO:root:Closing calibrator selector view
INFO:root:Launching calibrator with the following command: ['ros2', 'launch', '/home/kenzolobos/workspace/calibration_ws/install/sensor_calibration_manager/share/sensor_calibration_manager/launch/x2/tag_based_pnp_calibrator.launch.xml', 'camera_name:=camera6', 'view_only_ui:=true', 'calibration_pairs:=9', 'calibration_pairs_min_distance:=1.5']
[INFO] [launch]: All log files can be found below /home/kenzolobos/.ros/log/2024-08-30-14-34-49-741211-dpc2304008-2541214
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [image_transport_decompressor_node-1]: process started with pid [2541235]
[INFO] [lidartag_main-2]: process started with pid [2541237]
[INFO] [component_container-3]: process started with pid [2541239]
[INFO] [tag_based_pnp_calibrator-4]: process started with pid [2541241]
[INFO] [image_view_node.py-5]: process started with pid [2541243]
[INFO] [static_transform_publisher-6]: process started with pid [2541245]
[INFO] [rviz2-7]: process started with pid [2541247]
[static_transform_publisher-6] [WARN] [1724996090.048749686] []: Old-style arguments are deprecated; see --help for new-style arguments
[static_transform_publisher-6] [INFO] [1724996090.052088354] [tf_broadcaster]: Spinning until stopped - publishing transform
[static_transform_publisher-6] translation: ('0.000000', '0.000000', '0.000000')
[static_transform_publisher-6] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-6] from 'pandar_40p_front' to 'lidar_frame'
[component_container-3] [INFO] [1724996090.285041897] [apriltag.tag_container]: Load Library: /home/kenzolobos/workspace/calibration_ws/install/apriltag_ros/lib/libAprilTagNode.so
[component_container-3] [INFO] [1724996090.377948555] [apriltag.tag_container]: Found class: rclcpp_components::NodeFactoryTemplate<AprilTagNode>
[component_container-3] [INFO] [1724996090.377992405] [apriltag.tag_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<AprilTagNode>
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/apriltag' in container '/apriltag/tag_container'
[lidartag_main-2] [INFO] [1724996090.536203295] [lidartag.lidartag_main]: Use 40-beam LiDAR
[lidartag_main-2] 
[lidartag_main-2] [INFO] [1724996090.536807011] [lidartag.lidartag_main]: Use 12 threads
[lidartag_main-2] 
[lidartag_main-2] [INFO] [1724996090.537131537] [lidartag.lidartag_main]: depth_bound: 0.300000 
[lidartag_main-2] 
[lidartag_main-2] [INFO] [1724996090.537380093] [lidartag.lidartag_main]: payload_size_: 0.800000 
[lidartag_main-2] 
[lidartag_main-2] [INFO] [1724996090.537616544] [lidartag.lidartag_main]: horizontal_fov_: 360.000000 
[lidartag_main-2] 
[lidartag_main-2] [INFO] [1724996090.537863012] [lidartag.lidartag_main]: vertical_fov_: 40.000000 
[lidartag_main-2] 
[lidartag_main-2] [INFO] [1724996090.538097769] [lidartag.lidartag_main]: fine_cluster_threshold: 20 
[lidartag_main-2] 
[lidartag_main-2] [INFO] [1724996090.538324615] [lidartag.lidartag_main]: filling_gap_max_index: 400 
[lidartag_main-2] 
[lidartag_main-2] [INFO] [1724996090.538545769] [lidartag.lidartag_main]: points_threshold_factor: 1.500000 
[lidartag_main-2] 
[lidartag_main-2] [INFO] [1724996090.538763473] [lidartag.lidartag_main]: adaptive_thresholding_: 0 
[lidartag_main-2] 
[lidartag_main-2] [INFO] [1724996090.539005003] [lidartag.lidartag_main]: collect_dataset_: 0 
[lidartag_main-2] 
[lidartag_main-2] [INFO] [1724996090.539241428] [lidartag.lidartag_main]: decode_method: 3 
[lidartag_main-2] 
[lidartag_main-2] [INFO] [1724996090.539464769] [lidartag.lidartag_main]: linkage_hreshold_: 0.070711 
[lidartag_main-2] 
[lidartag_main-2] [INFO] [1724996090.539678667] [lidartag.lidartag_main]: ransac_threshold: 0.080000 
[lidartag_main-2] 
[lidartag_main-2] [INFO] [1724996090.539885787] [lidartag.lidartag_main]: num_accumulation_: 1 
[lidartag_main-2] 
[lidartag_main-2] =========================== 
[lidartag_main-2] use parameters from the launch file
[lidartag_main-2] =========================== 
[lidartag_main-2] 
[lidartag_main-2] 
[lidartag_main-2] ===== loading tag family ===== 
[lidartag_main-2] Preparing for tags: tag16h5
[lidartag_main-2] [INFO] [1724996090.540724303] [lidartag.lidartag_main]: ALL INITIALIZED!
[lidartag_main-2] [INFO] [1724996090.556962730] [lidartag.lidartag_main]: Waiting for pointcloud data
[rviz2-7] [INFO] [1724996090.746855734] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-7] [INFO] [1724996090.746953378] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-7] [INFO] [1724996090.811872140] [rviz2]: Stereo is NOT SUPPORTED
[lidartag_main-2] [INFO] [1724996102.082155473] [lidartag.lidartag_main]: Got the first pointcloud. Starting LidarTag
[lidartag_main-2] [INFO] [1724996102.082258006] [lidartag.lidartag_main]: Analyzing LiDAR Device

```

knzo25, 2024-09-10:
@ajay1606 
Was your problem solved?

ajay1606, 2024-09-17:
@knzo25 Sorry for late response, its working now, thank you so much for your feedback.

By the way, is there any ways to test manual/interactive extrinsic calibration for LiDAR and camera. Seems branch `feature/new_api_documentation` doesn't supports ?

Is there any alternative ways to calibrate Lidar and camera other than tag based solve pnp ? greatly appreciate your response.

knzo25, 2024-09-17:
@ajay1606 
This branch supports the interactive calibration (still PnP-based): https://github.com/tier4/CalibrationTools/pull/179
However, for real manual calibration you would need to publish tfs manually and tune them

ajay1606, 2024-09-17:
@knzo25 Thanks for the quick response, I will check and get back to you.
By the way, any chance to see stable calibration_tools branch, seems current main branch broken. Really missing this tool, I am using this tool since earlier version, but these days its quite hard to use all feature in single branch.

knzo25, 2024-09-17:
@ajay1606 
I feel you and as everyone else, would like to see this repository being more stable. Sadly, I am not involved in projects relating sensor calibration these days, yet we do not have anyone capable of handling it either.
That being said, I will force merge some of the PRs within this month.

knzo25, 2024-09-24:
The documentation was merged and how things should be more stable
https://github.com/tier4/CalibrationTools/pull/164

ajay1606, 2024-09-26:
@knzo25 Thank you so much for the update, now we are able to run `interactive calibration tool `. But having some issues with `initial_tf` publishing, we are still trying to fix that, soon will get back to you.

TimDing-aoi, 2024-09-27:
> @knzo25 Thank you so much for the update, now we are able to run `interactive calibration tool `. But having some issues with `initial_tf` publishing with custom data, we are still trying to fix that. Appreciate any inputs.
> 
> **Interactive calibration GUI** ![image](https://private-user-images.githubusercontent.com/29722843/371338555-7ba47d11-33a6-4bde-b155-a160b1321eba.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjc0NTIwOTUsIm5iZiI6MTcyNzQ1MTc5NSwicGF0aCI6Ii8yOTcyMjg0My8zNzEzMzg1NTUtN2JhNDdkMTEtMzNhNi00YmRlLWIxNTUtYTE2MGIxMzIxZWJhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTI3VDE1NDMxNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTMyMDBlYjc2YWE2YjcyOTIxYmQ0OGYwMDQ2MjFkMmUzYmE3MzlhZjc5NDliYmI3YjkyMWQzZWY3ZjJhMGQ0YjQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.dNjSbzpE6nyqxwuNpF8vwhK04sGrGgNR_cyFaXh-U8g)
> 
> **Output for tf_tree view_frames** ![image](https://private-user-images.githubusercontent.com/29722843/371338647-86c5eafd-43a7-4301-a3eb-889e6fbcefa8.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjc0NTIwOTUsIm5iZiI6MTcyNzQ1MTc5NSwicGF0aCI6Ii8yOTcyMjg0My8zNzEzMzg2NDctODZjNWVhZmQtNDNhNy00MzAxLWEzZWItODg5ZTZmYmNlZmE4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTI3VDE1NDMxNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTVlYTJiZDQ1NzljYTViOWJiZDlkYjdjZDhhZGNiOTE4OTY3YjQyNzZkNDFiNTY1YjY0OTQ3MzZjZGRlN2E3MTYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.wThYqoIyUGeKvK9Hwl0KPkgc-aKK9-Y1He0LFOkZrBA)
> 
> Any inputs ?

hello ajay, I am also doing lidar-camera calibration and I wonder how were  you able to launch this interactive calibration tool ?

ajay1606, 2024-09-27:
@TimDing-aoi Please check here : https://github.com/tier4/CalibrationTools/issues/185#issuecomment-2354415530
As @knzo25 mentioned, please use specific branch for the interactive tool.

knzo25, 2024-09-29:
@ajay1606 
The topic is diverging from the original issue (sample rosbag + camera-lidar). If that is so, please close the ticket and open a new one.

Regarding the interactive tool:
 - from your data I can not give you much feedback other than the frames are quite not standard. please check other camera-related ones
 - That tool is not really maintained actively since we do not use it in our projects and was developed as a means to evaluate some functionalities and correct some not calibrated data post-facto. We will not be able to dedicate time to give support to this particular tool as part of our OSS efforts

ajay1606, 2024-09-29:
@knzo25 Thank you so much for the support , will start new issue for additional queries if any.

closed 2024-09-29 by ajay1606