tier4/CalibrationTools #187: Issue with rosdep installation of Autoware calibration tools #183

jianhuaz, 2024-08-22:
ROS2 humble, autoware.universe

autoware.repos
...
universe/autoware.universe:
    type: git
    url: https://github.com/autowarefoundation/autoware.universe.git
    version: main
...

[image]

$ git checkout feature/new_api_documentation
$ vcs import src < calibration_tools_standalone.repos

$ rosdep install -y --from-paths src --ignore-src --rosdistro $ROS_DISTRO
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
ground_plane_calibrator: Cannot locate rosdep definition for [tier4_autoware_utils]
marker_radar_lidar_calibrator: Cannot locate rosdep definition for [tier4_autoware_utils]
sensor_calibration_tools: Cannot locate rosdep definition for [extrinsic_map_based_calibrator]
tag_based_pnp_calibrator: Cannot locate rosdep definition for [tier4_autoware_utils]
mapping_based_calibrator: Cannot locate rosdep definition for [tier4_autoware_utils]
tier4_ground_plane_utils: Cannot locate rosdep definition for [tier4_autoware_utils]
tag_based_sfm_calibrator: Cannot locate rosdep definition for [tier4_autoware_utils]
lidar_to_lidar_2d_calibrator: Cannot locate rosdep definition for [tier4_autoware_utils]
tunable_static_tf_broadcaster: Cannot locate rosdep definition for [tier4_autoware_utils]

or

$ vcs import src < calibration_tools_autoware.repos 
$ rosdep install -y --from-paths src --ignore-src --rosdistro $ROS_DISTRO
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
marker_radar_lidar_calibrator: Cannot locate rosdep definition for [tier4_autoware_utils]
tier4_ground_plane_utils: Cannot locate rosdep definition for [tier4_autoware_utils]
ground_plane_calibrator: Cannot locate rosdep definition for [tier4_autoware_utils]
tunable_static_tf_broadcaster: Cannot locate rosdep definition for [tier4_autoware_utils]
tag_based_sfm_calibrator: Cannot locate rosdep definition for [tier4_autoware_utils]
sensor_calibration_tools: Cannot locate rosdep definition for [extrinsic_map_based_calibrator]
lidar_to_lidar_2d_calibrator: Cannot locate rosdep definition for [tier4_autoware_utils]
tag_based_pnp_calibrator: Cannot locate rosdep definition for [tier4_autoware_utils]
mapping_based_calibrator: Cannot locate rosdep definition for [tier4_autoware_utils]

jianhuaz, 2024-08-22:
Install autoware.universe with autoware_universe_utils installed. When calibration_tools is installed, tier4_autoware_utils is referenced. How do I install calibration_tools in this error?__

knzo25, 2024-08-23:
Please see other issues related to this problem
For now, use the branch of this PR: https://github.com/tier4/CalibrationTools/pull/164

closed 2024-08-23 by knzo25

jianhuaz, 2024-08-23:
> Please see other issues related to this problem For now, use the branch of this PR: #164

I use the 'feature/new_api_documentation' branch, but I still report this error.

jianhuaz, 2024-08-23:
@knzo25

knzo25, 2024-08-26:
The instructions of that branch will not actually checkout that branch (since it expects to be merged)
You need to checkout the correct branch of the calibration tools after following the installation instructions. Please read related issues, it has been addressed before