tier4/CalibrationTools #198: lidar2ins(rtk)calibration

pedestrain123, 2024-10-02:
Hello, may I ask if there is an external parameter calibration algorithm for Lidar2ins? I see that you have open-source a ground-based lidar 2 baseline calibration method, but it can only calibrate roll and pitch, and cannot calibrate yaw and xy translation. Will the relevant calibration algorithms be open-source in the future? I recently looked at Apollo's calibration method, which first uses hand eye calibration to obtain initial extrinsic values, then concatenates the initial values with the INS pose to create a point cloud map, and finally uses GICP for extrinsic optimization. Unfortunately, Apollo does not have open-source code. Will you open source the lidar 2ins calibration code in the future?

knzo25, 2024-10-02:
Autoware at the moment does not fully use the INS pose (in the autoware tools repository there is a tool that compensates yaw) and as such we have not developed such a tool (it is in our eternal "would be nice to do" list).

If your objective is baseline to lidar calibration, we have the sfm tool that allows just that, but is more complex in the sense that requires more infrastructure from the users. 

Regarding OSS, so far everything we do at TIER IV in this area is planned to be open sourced

knzo25, 2024-10-09:
@pedestrain123 
Can I close this?

knzo25, 2024-10-21:
Closing this due to inactivity

closed 2024-10-21 by knzo25