tier4/CalibrationTools #209: [mapping_based_calibrator] Code Bug?

b06902126, 2024-11-08:
Hi, I'm using mapping_based_calibrator and tracing the code to see how you implement. Then, I'm confused in CalibrationMapper::checkKeyframeLost. 

[image]

It seems that you OR the same condition in if? Or there is a misunderstanding?

knzo25, 2024-11-08:
@b06902126 
Thanks ! It certainly looks like a bug. Will check it later and send a PR :pray:

knzo25, 2024-11-11:
Send a PR here: https://github.com/tier4/CalibrationTools/pull/211
Once that is merged, I will close this issue :pray:

knzo25, 2024-11-27:
The PR was merged so I will be closing this issue

closed 2024-11-27 by knzo25