tier4/CalibrationTools #215: [marker_radar_lidar_calibrator]

Sophie2216, 2024-11-25:
Hello!
Unfortunately, the marker radar lidar calibrator does not work for me. Hopefully, you can help me with the following problem:

I am trying to calibrate the lidar_top, which is a ouster_os2, with the lrr_front_ars548, which is a ARS548 3D radar. The initial TF tree is shown in the following screenshot:
[image]
For a better visualization, the TFs are located in the following way in 3D:
[image]
After the background extraction of the marker_radar_lidar_calibrator is finished, the corner reflector is placed into the environment and in the rviz2 visualization one can see the corner reflector as follows. Additionally, the blue dots are manually added to the marker_radar_lidar_calibrator view and showing the radar's detected objects according to the radar's topic.
[image]
A closer look is shown in the following two pictures, where it can be seen that the radars detected objects (blue) are not identical to the radar object_raw points (purple according your documentation of the marker_radar_lidar_calibrator). Several green marked lidar points (which are marker from the marker_radar_lidar_calibrator) also appear in the proximity of the corner reflector.
[image]
[image]
Since the marker_radar_lidar_calibrator finds lidar (green) and radar (purple) dots, the "Add lidar-radar pair" button was able to be pressed. Unfortunately, the pair is not established and the console has the following output:
`[marker_radar_lidar_calibrator-2] [INFO] [1732291567.283530748] [marker_radar_lidar_calibrator]: lidarCallback
[marker_radar_lidar_calibrator-2] [INFO] [1732291567.283549330] [marker_radar_lidar_calibrator]: There were no tracks
[marker_radar_lidar_calibrator-2] [INFO] [1732291567.336945966] [marker_radar_lidar_calibrator]: lidarCallback
[marker_radar_lidar_calibrator-2] [INFO] [1732291567.339657780] [marker_radar_lidar_calibrator]: Extracting foreground
[marker_radar_lidar_calibrator-2] [INFO] [1732291567.339695228] [marker_radar_lidar_calibrator]: 	 initial points: 131072
[marker_radar_lidar_calibrator-2] [INFO] [1732291567.340580046] [marker_radar_lidar_calibrator]: 	 cropped points: 129459
[marker_radar_lidar_calibrator-2] [INFO] [1732291567.343981659] [marker_radar_lidar_calibrator]: 	 voxel filtered points: 285
[marker_radar_lidar_calibrator-2] [INFO] [1732291567.344404879] [marker_radar_lidar_calibrator]: 	 tree filtered points: 12
[marker_radar_lidar_calibrator-2] [INFO] [1732291567.480623977] [marker_radar_lidar_calibrator]: 	 ransac filtered points: 12
[marker_radar_lidar_calibrator-2] [INFO] [1732291567.480723653] [marker_radar_lidar_calibrator]: Cluster extraction input size: 12
[marker_radar_lidar_calibrator-2] [INFO] [1732291567.480730139] [marker_radar_lidar_calibrator]: 	 found cluster of size: 12
[marker_radar_lidar_calibrator-2] [INFO] [1732291567.480738445] [marker_radar_lidar_calibrator]: Extracting lidar reflectors from clusters
[marker_radar_lidar_calibrator-2] [INFO] [1732291567.480745983] [marker_radar_lidar_calibrator]: Colored clusters size=12
[marker_radar_lidar_calibrator-2] [INFO] [1732291567.481355589] [marker_radar_lidar_calibrator]: Extracting foreground
[marker_radar_lidar_calibrator-2] [INFO] [1732291567.481374676] [marker_radar_lidar_calibrator]: 	 initial points: 200
[marker_radar_lidar_calibrator-2] [INFO] [1732291567.481387904] [marker_radar_lidar_calibrator]: 	 cropped points: 135
[marker_radar_lidar_calibrator-2] [INFO] [1732291567.481406840] [marker_radar_lidar_calibrator]: 	 voxel filtered points: 127
[marker_radar_lidar_calibrator-2] [INFO] [1732291567.481496107] [marker_radar_lidar_calibrator]: 	 tree filtered points: 70
[marker_radar_lidar_calibrator-2] [INFO] [1732291567.481580595] [marker_radar_lidar_calibrator]: Cluster extraction input size: 70
[marker_radar_lidar_calibrator-2] [INFO] [1732291567.481588946] [marker_radar_lidar_calibrator]: 	 found cluster of size: 4
[marker_radar_lidar_calibrator-2] [INFO] [1732291567.481593272] [marker_radar_lidar_calibrator]: 	 found cluster of size: 4
[marker_radar_lidar_calibrator-2] [INFO] [1732291567.481597285] [marker_radar_lidar_calibrator]: 	 found cluster of size: 4
[marker_radar_lidar_calibrator-2] [INFO] [1732291567.481600920] [marker_radar_lidar_calibrator]: 	 found cluster of size: 4
[marker_radar_lidar_calibrator-2] [INFO] [1732291567.481604756] [marker_radar_lidar_calibrator]: 	 found cluster of size: 4
[marker_radar_lidar_calibrator-2] [INFO] [1732291567.481608270] [marker_radar_lidar_calibrator]: 	 found cluster of size: 4
[marker_radar_lidar_calibrator-2] [INFO] [1732291567.481611837] [marker_radar_lidar_calibrator]: 	 found cluster of size: 4
[marker_radar_lidar_calibrator-2] [INFO] [1732291567.481615115] [marker_radar_lidar_calibrator]: 	 found cluster of size: 4
[marker_radar_lidar_calibrator-2] [INFO] [1732291567.481618768] [marker_radar_lidar_calibrator]: 	 found cluster of size: 4
[marker_radar_lidar_calibrator-2] [INFO] [1732291567.481622034] [marker_radar_lidar_calibrator]: 	 found cluster of size: 4
[marker_radar_lidar_calibrator-2] [INFO] [1732291567.481625137] [marker_radar_lidar_calibrator]: 	 found cluster of size: 4
[marker_radar_lidar_calibrator-2] [INFO] [1732291567.481628448] [marker_radar_lidar_calibrator]: 	 found cluster of size: 4
[marker_radar_lidar_calibrator-2] [INFO] [1732291567.481631796] [marker_radar_lidar_calibrator]: 	 found cluster of size: 4
[marker_radar_lidar_calibrator-2] [INFO] [1732291567.481634958] [marker_radar_lidar_calibrator]: 	 found cluster of size: 4
[marker_radar_lidar_calibrator-2] [INFO] [1732291567.481638471] [marker_radar_lidar_calibrator]: 	 found cluster of size: 4
[marker_radar_lidar_calibrator-2] [INFO] [1732291567.481642061] [marker_radar_lidar_calibrator]: 	 found cluster of size: 4
[marker_radar_lidar_calibrator-2] [INFO] [1732291567.481645473] [marker_radar_lidar_calibrator]: 	 found cluster of size: 4
[marker_radar_lidar_calibrator-2] [INFO] [1732291567.481649190] [marker_radar_lidar_calibrator]: 	 found cluster of size: 1
[marker_radar_lidar_calibrator-2] [INFO] [1732291567.481652909] [marker_radar_lidar_calibrator]: 	 found cluster of size: 1
[marker_radar_lidar_calibrator-2] [INFO] [1732291567.481659360] [marker_radar_lidar_calibrator]: Extracting radar reflectors from clusters
[marker_radar_lidar_calibrator-2] [INFO] [1732291567.481663864] [marker_radar_lidar_calibrator]: 	 Radar reflector id=0 size=4 center: x=48.80 y=31.76 z=1.38
[marker_radar_lidar_calibrator-2] [INFO] [1732291567.481670877] [marker_radar_lidar_calibrator]: 	 Radar reflector id=1 size=4 center: x=46.23 y=-12.88 z=1.37
[marker_radar_lidar_calibrator-2] [INFO] [1732291567.481676019] [marker_radar_lidar_calibrator]: 	 Radar reflector id=2 size=4 center: x=46.57 y=-0.98 z=0.97
[marker_radar_lidar_calibrator-2] [INFO] [1732291567.481694995] [marker_radar_lidar_calibrator]: 	 Radar reflector id=3 size=4 center: x=43.78 y=0.54 z=0.74
[marker_radar_lidar_calibrator-2] [INFO] [1732291567.481700286] [marker_radar_lidar_calibrator]: 	 Radar reflector id=4 size=4 center: x=43.71 y=31.40 z=1.52
[marker_radar_lidar_calibrator-2] [INFO] [1732291567.481705257] [marker_radar_lidar_calibrator]: 	 Radar reflector id=5 size=4 center: x=38.28 y=-7.61 z=0.98
[marker_radar_lidar_calibrator-2] [INFO] [1732291567.481710326] [marker_radar_lidar_calibrator]: 	 Radar reflector id=6 size=4 center: x=32.35 y=13.40 z=1.68
[marker_radar_lidar_calibrator-2] [INFO] [1732291567.481715132] [marker_radar_lidar_calibrator]: 	 Radar reflector id=7 size=4 center: x=30.22 y=-4.73 z=0.95
[marker_radar_lidar_calibrator-2] [INFO] [1732291567.481719935] [marker_radar_lidar_calibrator]: 	 Radar reflector id=8 size=4 center: x=24.12 y=-3.04 z=0.46
[marker_radar_lidar_calibrator-2] [INFO] [1732291567.481724275] [marker_radar_lidar_calibrator]: 	 Radar reflector id=9 size=4 center: x=20.25 y=-3.39 z=0.05
[marker_radar_lidar_calibrator-2] [INFO] [1732291567.481728954] [marker_radar_lidar_calibrator]: 	 Radar reflector id=10 size=4 center: x=13.09 y=-10.34 z=1.01
[marker_radar_lidar_calibrator-2] [INFO] [1732291567.481733534] [marker_radar_lidar_calibrator]: 	 Radar reflector id=11 size=4 center: x=10.21 y=-0.30 z=-0.02
[marker_radar_lidar_calibrator-2] [INFO] [1732291567.481738467] [marker_radar_lidar_calibrator]: 	 Radar reflector id=12 size=4 center: x=9.80 y=-2.76 z=0.47
[marker_radar_lidar_calibrator-2] [INFO] [1732291567.481743346] [marker_radar_lidar_calibrator]: 	 Radar reflector id=13 size=4 center: x=10.57 y=-5.40 z=0.13
[... cut ...]
[marker_radar_lidar_calibrator-2] [INFO] [1732291567.481766340] [marker_radar_lidar_calibrator]: 	 Radar reflector id=18 size=1 center: x=15.98 y=-15.70 z=0.37
[marker_radar_lidar_calibrator-2] [INFO] [1732291567.482007915] [marker_radar_lidar_calibrator]: Lidar detections: 0 Radar detections: 19 Matches: 0 Active tracks; 0 Converged tracks: 0
`
So it seems that radar detections are found but no lidar detections according to the console output although in the rviz2 visualization there are green marked lidar points.

So, why is it not possible to add the lidar-radar pair?

knzo25, 2024-11-26:
@Sophie2216 
Hi, thank for using our tools.

Regarding the ARS548, we support it in this [branch](https://github.com/tier4/CalibrationTools/pull/180). Are you using it?
In a little unrelated topic, are you using our driver for that device?

Until the related PR is merged it is difficult to give proper support. That being said, if you provide us some rosbag, we can at least have a look a it !

Sophie2216, 2024-11-27:
@knzo25 
Hi,
Currently, we are using the tier4/universe. So, thanks for redirecting us to the "feature/marker_radar_lidar_calibrator_support_radars_and_transformation_algorithms" branch.
Yes, we are using your driver for that device.

I would like to send you the rosbag via email. Would this be possible?

knzo25, 2024-11-27:
Thanks for the confirmation, the before mentioned PR used our driver so I can not really give any sort of guarantees otherwise.
My mail is kenzo.lobos at tier4.jp

knzo25, 2024-12-11:
@Sophie2216 
Sorry, I have been away for an experiment. As soon as I finish catching up with work I will take a look

Sophie2216, 2024-12-13:
@knzo25 
No problem!

We have tried to do the lidar-radar calibration by using the "feature/marker_radar_lidar_calibrator_support_radars_and_transformation_algorithms" branch. Unfortunately, it didn't work. The default project was selected in combination with the marker_radar_lidar_calibrator and the following variables were inserted in the launcher configuration:
[image]

We thought that the radar's scan_raw topic may fit best from the following possible topics of the ARS548 radar:
/sensing/radar/lrr_front_ars548/detection_points
/sensing/radar/lrr_front_ars548/detections
/sensing/radar/lrr_front_ars548/marker_array
/sensing/radar/lrr_front_ars548/nebula_packets
/sensing/radar/lrr_front_ars548/object_points
/sensing/radar/lrr_front_ars548/objects
/sensing/radar/lrr_front_ars548/objects_raw
/sensing/radar/lrr_front_ars548/objects_raw_filtered
/sensing/radar/lrr_front_ars548/output/radar_detected_objects
/sensing/radar/lrr_front_ars548/output/radar_tracked_objects
/sensing/radar/lrr_front_ars548/scan_raw

The sensor_calibration_manager window, RViz, and the metrics plotter appeared, but the 'Calibrate' button was not available and in the terminal the following lines showed up:
{ade}-adm@innb0:~/WARE/calibration_tools_ws$ros2 run sensor_calibration_manager sensor_calibration_manager 
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-ware'
INFO:root:on_selected_calibrator: project_name=default_project calibrator_name=marker_radar_lidar_calibrator
INFO:root:Reading xml from: /home/ware/WARE/calibration_tools_ws/install/sensor_calibration_manager/share/sensor_calibration_manager/launch/default_project/marker_radar_lidar_calibrator.launch.xml
INFO:root:Closing calibrator selector view
^[[AINFO:root:Launching calibrator with the following command: ['ros2', 'launch', '/home/ware/WARE/calibration_tools_ws/install/sensor_calibration_manager/share/sensor_calibration_manager/launch/default_project/marker_radar_lidar_calibrator.launch.xml', 'radar_optimization_frame:=base_link', 'radar_frame:=lrr_front_ars548', 'lidar_frame:=lidar_top/os_lidar', 'msg_type:=radar_scan', 'transformation_type:=svd_3d', 'input_radar_msg:=/sensing/radar/lrr_front_ars548/scan_raw', 'input_lidar_pointcloud:=/sensing/lidar/top/points', 'rviz:=true']
[INFO] [launch]: All log files can be found below /home/ware/.ros/log/2024-12-13-14-15-05-821125-ware2-11976
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [static_transform_publisher-1]: process started with pid [11998]
[INFO] [marker_radar_lidar_calibrator-2]: process started with pid [12000]
[INFO] [calibrator_ui_node.py-3]: process started with pid [12002]
[INFO] [metrics_plotter_node.py-4]: process started with pid [12004]
[INFO] [rviz2-5]: process started with pid [12006]
[static_transform_publisher-1] [WARN] [1734095706.152362657] []: Old-style arguments are deprecated; see --help for new-style arguments
[static_transform_publisher-1] [INFO] [1734095706.162631928] [lidar_broadcaster]: Spinning until stopped - publishing transform
[static_transform_publisher-1] translation: ('0.000000', '0.000000', '0.000000')
[static_transform_publisher-1] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-1] from 'lrr_front_ars548' to 'radar_frame'
[marker_radar_lidar_calibrator-2] ERROR: flag 'logtostderr' was defined more than once (in files './src/logging.cc' and '/home/ware/WARE/autoware/src/universe/external/glog/src/logging.cc').
[ERROR] [marker_radar_lidar_calibrator-2]: process has died [pid 12000, exit code 1, cmd '/home/ware/WARE/calibration_tools_ws/install/marker_radar_lidar_calibrator/lib/marker_radar_lidar_calibrator/marker_radar_lidar_calibrator --ros-args -r __node:=marker_radar_lidar_calibrator -r __ns:=/ --params-file /tmp/launch_params_b5w3go4o --params-file /tmp/launch_params_6kcf_vkz --params-file /tmp/launch_params_qb1f3nna --params-file /tmp/launch_params_nhoc1w32 --params-file /tmp/launch_params_d39u93di --params-file /tmp/launch_params_l3uild_r --params-file /tmp/launch_params_6mek7k99 --params-file /tmp/launch_params___den0ar --params-file /tmp/launch_params_edolvfxj --params-file /tmp/launch_params_imn0r8is --params-file /tmp/launch_params_u06dit3h --params-file /tmp/launch_params_hpoke5d2 --params-file /tmp/launch_params_o3fuzqbh --params-file /tmp/launch_params_g4_rq1wu --params-file /tmp/launch_params_5v02qdqq --params-file /tmp/launch_params_xirtwwkv --params-file /tmp/launch_params_s0dy57ls -r extrinsic_calibration:=calibrate_radar_lidar -r input_lidar_pointcloud:=/sensing/lidar/top/points -r input_radar_msg:=/sensing/radar/lrr_front_ars548/scan_raw'].

The following xml-file (I had to upload it as a txt-file since xml is not a supported file format for uploading) was used for launching the ARS548:
[lrr_front_ars548.launch.txt](https://github.com/user-attachments/files/18127365/lrr_front_ars548.launch.txt)

I hope that you can help with this amount of information to solve the issue such that we can calibrate with the marker_radar_lidar_calibrator.

Thanks for your help!

Sophie2216, 2024-12-17:
Hi!
Do you have any idea why the marker_radar_lidar_calibrator from branch "feature/marker_radar_lidar_calibrator_support_radars_and_transformation_algorithms" is not working for us?

Sophie2216, 2025-01-09:
Hello @knzo25 ,
Do you already have an idea why this calibration is failing for me and how this could possibly be resolved? 
I would highly appreciate receiving an answer as soon as possible.

knzo25, 2025-01-09:
@Sophie2216 
I will try it now (finally got a breather while I crunch some data), but please remember that this repository is an OSS effort.

Sophie2216, 2025-01-10:
@knzo25 
Thanks for the quick reply! Very appreciated!

We added a frame lrr_front_ars548_footprint with xyz:=0 0 0 and rpy:=0 0 0, which has a parent the frame lrr_front_ars548.

First, we tried the "feature/marker_radar_lidar_calibrator_support_radars_and_transformation_algorithms" branch with the svd_2d algorithm. The recommended /sensing/radar/lrr_front_ars548/detections with the msg_type:=radar_cloud was used and led to the same error in the terminal als shown in my previous comment on Dec, 13th 2024.

Thus, we pulled the "tier4/universe" branch. When we used this branch, we wanted to use the detections as you suggested, but it was not possible since in the terminal the marker_radar_lidar_calibrator has written that the type RadarTracks is required. Unfortunately, /sensing/radar/lrr_front_ars548/detections has the type PointCloud. Thus, we tried it with /sensing/radar/lrr_front_ars548/objects_raw. With this topic it worked  in the end such that we were able to register the radar-lidar pairs. **BUT** the /sensing/radar/lrr_front_ars548/objects_raw topic has as the header.frame_id the base_link. This does not calibrate the tf links from base_link to lrr_front_ars548.

Additionally, when we added the radar-lidar pairs and clicked "Send calibration", an error appeared in the sensor_calibration_manager window which said something like "Error. tf is not included" such that calibration was not able to be saved.

How can we use the detections topic for calibration?
Why does the feature/marker_radar_lidar_calibrator_support_radars_and_transformation_algorithms branch throw that error?

knzo25, 2025-01-15:
I do not remember the specifics of the radar calibration development branch, for that you would need to reach its author.

`radar_cloud` should use the radar detections though (you can verify it through this and the driver repos)
`Error. tf is not included` seems to be an issue with the integratin of the `sensor_calibration manager` (either the launch or python counterpart)

Sophie2216, 2025-01-15:
@vividf I have seen that you are an author of the branch 'feature/marker_radar_lidar_calibrator_support_radars_and_transformation_algorithms'. I need your expertise for solving my issue with the marker_radar_lidar_calibrator. I would highly appreciate receiving an answer as soon as possible. Please can you help me to find out what is the reason that the process dies as shown in my previous comment on Dec, 13th 2024 and how can I resolve the issue?

We would like to use this branch since it seems to be the only way to use the /sensing/radar/lrr_front_ars548/detections topic of the Radar ARS548, which has the type PointCloud. In the comment on January, 10th 2025, a better insight is given why we have to use this topic instead of using /sensing/radar/lrr_front_ars548/objects_raw, which would work already with the branch 'tier4/universe'.

vividf, 2025-01-15:
@Sophie2216 Hi,

Could you please test using this [branch](https://github.com/vividf/CalibrationTools/tree/feature/4d_radar_lidar_calibration_tool)? It contains the latest updates for supporting the ARS548.

For ARS548, please use either `radar_cloud` or `radar_scan` instead of `radar_track`. Based on the information I received earlier, the `radar_track` frame ID is set to `base_link` instead of `radar_frame`.

Sophie2216, 2025-01-20:
@vividf Hi,
I was able to try the branch, which you recommended. The following parameters were used:
radar_optimization_frame = base_link     (Since I saw that you used base_link here for ARS548, I used it too)
radar_frame = lrr_front_ars548
lidar_frame = lidar_top/os_lidar
msg_type = radar_tracks e= radar_cloud
transformation_type = svd_3d
input_radar_msg = /sensing/radar/lrr_front_ars548/detection_points
input_lidar_pointcloud = /sensing/lidar/top/points

Unfortunately, I was not able to add a lidar-raidar pair due to the flickering although I tried clicking multiple times the available button "Add lidar-radar pair". In the follwing video, the problem is shown by the projected foreground points of lidar and radar:

[image]

How can I fix this problem?

vividf, 2025-01-21:
@Sophie2216 
Hi, could you check the frequency of your lidar and radar topics?
I didn't have the scenario you met before.

Sophie2216, 2025-01-21:
@vividf 
Hi, is there a special requirement for the frequency of the lidar and radar topics except the fact that they should be synchronized?

Also, wouldn't this problem have already occurred in the tier4/universe branch (svd_2d algorithm)? There I didn't have a problem to add a pair since there was no flickering.

vividf, 2025-01-21:
@Sophie2216 Could you provide more details about your video?

What playback speed are you using for your ROS bag?
Please record the video using Ubuntu's screencast feature for better clarity.
Additionally, could you elaborate on your statement:
"Also, wouldn't this problem have already occurred in the tier4/universe branch (svd_2d algorithm)? There, I didn’t have an issue adding a pair since there was no flickering."
Please provide supporting evidence for this observation.

As I mentioned before, please share the frequency of your topics. Based on your video, it appears that the point cloud updates much slower compared to the radar topics.

Sophie2216, 2025-01-23:
@vividf I would like to provide you a rosbag and the screencast video but unfortunately here it is not possible to upload it since it is too big. Could you please give me your email?

About the freqencies of radar and lidar, the following rates of the topics have been measured:

[image]

Sophie2216, 2025-01-28:
@vividf Do you think that it is a problem when the frequency of /sensing/radar/lrr_front_ars548/detection_points is about 20 Hz and the frequency of /sensing/lidar/top/points is about 10 Hz or do you think there is another reason for this problem?

Sophie2216, 2025-02-07:
Hi @vividf !
So, we tried the marker_radar_lidar_calibrator from your branch 'feature/4d_radar_lidar_calibration_tool' with the latest commit on January, 23rd 2025 with the following settings. The pointcloud of the lidar has been cropped by a self-created ROS node, since we have experienced that if it isn't cropped there are two many points and no clustering of the reflector is achieved. Also, we set the frame of the lidar to base_link by using the same ROS node.

[image]

As it can be seen in the following picture, the radar point and the lidar point is matched:

[image]

or in the following video:

https://github.com/user-attachments/assets/fb9e2ac0-385e-4851-aafb-b759136d1f82

After pressing the 'Add lidar-radar pair' button, the console reveals the following lines. It is shown that the pair is matched but somehow it is not possible that the pair converges such that it is registered and saved.
```
[marker_radar_lidar_calibrator-2] [INFO] [1738933071.732856020] [marker_radar_lidar_calibrator]: lidarCallback
[marker_radar_lidar_calibrator-2] [INFO] [1738933071.732949887] [marker_radar_lidar_calibrator]: Extracting foreground
[marker_radar_lidar_calibrator-2] [INFO] [1738933071.732958758] [marker_radar_lidar_calibrator]: 	 initial points: 7
[marker_radar_lidar_calibrator-2] [INFO] [1738933071.732966834] [marker_radar_lidar_calibrator]: 	 cropped points: 7
[marker_radar_lidar_calibrator-2] [INFO] [1738933071.732975256] [marker_radar_lidar_calibrator]: 	 voxel filtered points: 2
[marker_radar_lidar_calibrator-2] [INFO] [1738933071.732998035] [marker_radar_lidar_calibrator]: 	 tree filtered points: 2
[marker_radar_lidar_calibrator-2] [INFO] [1738933071.733041411] [marker_radar_lidar_calibrator]: Cluster extraction input size: 2
[marker_radar_lidar_calibrator-2] [INFO] [1738933071.733049249] [marker_radar_lidar_calibrator]: 	 found cluster of size: 1
[marker_radar_lidar_calibrator-2] [INFO] [1738933071.733054330] [marker_radar_lidar_calibrator]: 	 found cluster of size: 1
[marker_radar_lidar_calibrator-2] [INFO] [1738933071.733062527] [marker_radar_lidar_calibrator]: Extracting radar reflectors from clusters
[marker_radar_lidar_calibrator-2] [INFO] [1738933071.733069710] [marker_radar_lidar_calibrator]: 	 Radar reflector id=0 size=1 center: x=5.92 y=-0.14 z=0.37
[marker_radar_lidar_calibrator-2] [INFO] [1738933071.733079355] [marker_radar_lidar_calibrator]: 	 Radar reflector id=1 size=1 center: x=15.83 y=4.28 z=0.21
[marker_radar_lidar_calibrator-2] [INFO] [1738933071.733157590] [marker_radar_lidar_calibrator]: pre lidar_pointcloud_ptr=1015
[marker_radar_lidar_calibrator-2] [WARN] [1738933071.733164265] [marker_radar_lidar_calibrator]: crop box parameters=-20.000000 | -5.000000 | -5.000000
[marker_radar_lidar_calibrator-2] [WARN] [1738933071.733171690] [marker_radar_lidar_calibrator]: crop box parameters=20.000000 | 5.000000 | 5.000000
[marker_radar_lidar_calibrator-2] [INFO] [1738933071.733184747] [marker_radar_lidar_calibrator]: lidar_pointcloud_ptr=982
[marker_radar_lidar_calibrator-2] [INFO] [1738933071.733190253] [marker_radar_lidar_calibrator]: Extracting foreground
[marker_radar_lidar_calibrator-2] [INFO] [1738933071.733219325] [marker_radar_lidar_calibrator]: 	 initial points: 982
[marker_radar_lidar_calibrator-2] [INFO] [1738933071.733228548] [marker_radar_lidar_calibrator]: 	 cropped points: 982
[marker_radar_lidar_calibrator-2] [INFO] [1738933071.733280297] [marker_radar_lidar_calibrator]: 	 voxel filtered points: 47
[marker_radar_lidar_calibrator-2] [INFO] [1738933071.733320803] [marker_radar_lidar_calibrator]: 	 tree filtered points: 39
[marker_radar_lidar_calibrator-2] [INFO] [1738933071.733419751] [marker_radar_lidar_calibrator]: 	 ransac filtered points: 39
[marker_radar_lidar_calibrator-2] [INFO] [1738933071.733513083] [marker_radar_lidar_calibrator]: Cluster extraction input size: 39
[marker_radar_lidar_calibrator-2] [INFO] [1738933071.733518812] [marker_radar_lidar_calibrator]: 	 found cluster of size: 39
[marker_radar_lidar_calibrator-2] [INFO] [1738933071.733526198] [marker_radar_lidar_calibrator]: Extracting lidar reflectors from clusters
[marker_radar_lidar_calibrator-2] [INFO] [1738933071.733533371] [marker_radar_lidar_calibrator]: 	 found cluster
[marker_radar_lidar_calibrator-2] [INFO] [1738933071.733544550] [marker_radar_lidar_calibrator]: 	 Lidar reflector id=0 size=6 center: x=9.45 y=0.05 z=0.97
[marker_radar_lidar_calibrator-2] [INFO] [1738933071.733554317] [marker_radar_lidar_calibrator]: Colored clusters size=39
[marker_radar_lidar_calibrator-2] [INFO] [1738933071.733609174] [marker_radar_lidar_calibrator]: Lidar reflectors in radar coordinate system (using the initial transformation)
[marker_radar_lidar_calibrator-2] [INFO] [1738933071.733614890] [marker_radar_lidar_calibrator]: 	 Lidar reflector (rcs) id=0 size=1 center: x=5.71 y=-0.18 z=0.33
[marker_radar_lidar_calibrator-2] [INFO] [1738933071.733621463] [marker_radar_lidar_calibrator]: Matching each lidar detections to its closest radar detection
[marker_radar_lidar_calibrator-2] [INFO] [1738933071.733626221] [marker_radar_lidar_calibrator]: 	Closest radar to lidar=0 is 0 with distance 0.214567
[marker_radar_lidar_calibrator-2] [INFO] [1738933071.733631641] [marker_radar_lidar_calibrator]: Matching each radar detections to its closest lidar detection
[marker_radar_lidar_calibrator-2] [INFO] [1738933071.733636569] [marker_radar_lidar_calibrator]: 	Closest lidar to radar=0 is 0 with distance 0.214567
[... cut ...]
[marker_radar_lidar_calibrator-2] [INFO] [1738933071.776149480] [marker_radar_lidar_calibrator]: lidar_index = 0 / 1
[marker_radar_lidar_calibrator-2] [INFO] [1738933071.776154412] [marker_radar_lidar_calibrator]: closest_radar_index = 0 / 1
[marker_radar_lidar_calibrator-2] [INFO] [1738933071.776243302] [marker_radar_lidar_calibrator]: Lidar detections: 1 Radar detections: 1 Matches: 1 Converged tracks: 0
```
Why is it not possible to let this matched pair converge such that it is a registered converged ridar-lidar pair?