tier4/CalibrationTools #232: Docker installation fails

fnobis, 2025-02-10:
Trying to install the calibration tools via docker with

`DOCKER_BUILDKIT=1 docker build --ssh default -t ghcr.io/tier4/sensor-calibration-tools:2.0 -f docker/Dockerfile ..`

however the process fails:

```
=> ERROR [ 8/11] RUN rosdep install -y --from-paths `colcon list --packages-up-to sensor_calibration_tools -p` --ignore-src                                                                                                                                                                                                                                                                                                                  6.6s 
------                                                                                                                                                                                                                                                                                                                                                                                                                                             
 > [ 8/11] RUN rosdep install -y --from-paths `colcon list --packages-up-to sensor_calibration_tools -p` --ignore-src:                                                                                                                                                                                                                                                                                                                             
6.469 ERROR: the following packages/stacks could not have their rosdep keys resolved                                                                                                                                                                                                                                                                                                                                                               
6.469 to system dependencies:                                                                                                                                                                                                                                                                                                                                                                                                                      
6.469 mapping_based_calibrator: Cannot locate rosdep definition for [autoware_kalman_filter]                                                                                                                                                                                                                                                                                                                                                       
6.469 marker_radar_lidar_calibrator: Cannot locate rosdep definition for [autoware_kalman_filter]                                                                                                                                                                                                                                                                                                                                                  
6.469 lidar_to_lidar_2d_calibrator: Cannot locate rosdep definition for [autoware_kalman_filter]
6.469 ground_plane_calibrator: Cannot locate rosdep definition for [autoware_kalman_filter]
```

I see that there is a new branch `chore/core_migration` which mentions this problem, but after checking out that branch, and running with "--no-cache" option, the problem still occurs. Is there anything else I need to modify?

knzo25, 2025-02-12:
@fnobis 
Thanks for your report. I have not used the tool lately so I have't checked the status of the docker build.
Will try to have it worked by the end of this week

fnobis, 2025-02-13:
Hey @knzo25, 

thanks for fixing this. Works great now!

I just noticed that my confusion was coming from the fact that the `calibration_tools_standalone.repos` file was correctly updated in the branch in the repo. But the docker file downloads that file again from the internet instead of using the updated local repo one, probably because of the folder structure and the docker buid system.

Dockerfile l.16:
```
RUN wget https://raw.githubusercontent.com/tier4/CalibrationTools/tier4/universe/calibration_tools_standalone.repos --directory-prefix=/workspace
```

closed 2025-02-13 by fnobis