tier4/CalibrationTools #234: [tag based pnp calibrator] big reprojection error

Sophie2216, 2025-02-19:
Hello!
We got wrong extrinsic parameter, which are about 10cm away from the correct values (translation) and have a wrong orientation (rotation). Thus, we had a closer look on what possibly goes wrong. We saw that we have a really large reprojection error in comparison to the screenshots from your documentation of the tag_based_pnp_calibrator. Our setup includes a Ouster OS2 Lidar and a Tier IV C1camera.
We have used the calibrator with the option use_rectified_image:=false. We got a reprojection error of 22.56 as it can be seen in the following screenshots, which were taken after the calibration finished.
[image]
[image]
The following lines appeared in the console:
```
[tag_based_pnp_calibrator-3] [INFO] [1739979192.341200411] [tag_based_pnp_calibrator_node]: Partial calibration results (9/9 pairs):
[tag_based_pnp_calibrator-3] [INFO] [1739979192.341258921] [tag_based_pnp_calibrator_node]:     Initial reprojection error=30.06
[tag_based_pnp_calibrator-3] [INFO] [1739979192.341278020] [tag_based_pnp_calibrator_node]:     Current reprojection error=20.54
[tag_based_pnp_calibrator-3] [INFO] [1739979192.341287684] [tag_based_pnp_calibrator_node]:     Filtered reprojection error=20.40
```

Then, we thought that maybe we get a better reprojection error if we set the option use_rectified_image:=true since this camera has a preprocessing such that the image is mainly rectified. With this settings, we achieved a better but still to huge reprojection error of 18.64. The following screenshots were taken after the calibration finished.
[image]
[image]
[image]

The following settings were identical in both tries, where we had 9 registered calibration pairs:
camera_optical_link_frame := camera_front_left
camera_info_topic := /sensing/camera/front_left/camera_info
image_decompressed_topic := /sensing/camera/front_left/image_raw
image_compressed_topic := /sensing/camera/front_left/image_raw/compressed
lidar_frame := lidar_top
pointcloud_topic := /sensing/lidar/top/points

Why do we have such a big reprojection error?

knzo25, 2025-02-20:
@Sophie2216 
Thanks for bringing up this to us.
This is one of our first tools and probably the most mature of them so I am quite interested in the reasons behind this.
If I can get data to reproduce it I can probably give you a verdict within the week (I also need to fix an OSS launcher so I would do both rather soon).

That being said, potential causes are:
 - The UI shows that lidar scans are vertical, nor horizontal. Probably some frames are mis-configured.
 - We have has very bad reprojection errors when the intrinsics are not good or they belong to another lens (e.g, when using the intrinsics of a C1-6X on a C1-9X)

Sophie2216, 2025-02-20:
@knzo25 
That would be great if you can give a verdict within a week. Many thanks in advance!
Under the following [link](https://v2c2-my.sharepoint.com/:f:/g/personal/sophie_weikovich_v2c2_at/Eoew_-aMMu9LkyXqazVQgBcBguGBPrIGlv8mJMjSQoveuQ?e=CCzR02) you can find the rosbag, which we recorded over the whole calibration process. During this calibration process we also took the following screenshots: 
[image]
[image]
[image]
[image]
I hope that this data are sufficient to get to the bottom of the problem.

knzo25, 2025-02-25:
@Sophie2216 
I had some time to look at it today.
The weird vertical patterns is just an artifact. If you set to subsampling to 1 it look ok (I had never seen that happen with our sensors, which have a less organized pattern).

The high error seems to be that your intrinsics are not good enough. For reference, check the following visualization and error when you have "better" intrinsics.

[image]

Sophie2216, 2025-02-25:
@knzo25 
Thank you! I tried it today with the intrinsic parameters which I got by calibrating the camera with the intrinsic_camera_calibrator and you are absolutely right because the reprojection error has reduced to a value below 3.

Now, I also know why I have such a big difference between the real extrinsic parameters and the extrinsic parameters gained from the tag_based_pnp_calibrator. This leads me to a new problem: From the tag_based_pnp_calibrator I got extrinsic parameters from camera_front_left to lidar_top/os_lidar. Unfortunately, I need for my tf tree the opposite direction of the extrinsic parameters namely from the lidar_top/os_lidar to camera_front_left. I have seen that the extrinsic parameters from tag_based_pnp_calibrator are correct, when I view them in Rviz2 by publishing the tf. 

Thus, my new question is: Do you know how can I retrieve the extrinsic parameters for lidar_top/os_lidar to camera_front_left when I have the extrinsic parameters from camera_front_left to lidar_top/os_lidar?

As a reference the result from the tag_based_pnp_calibrator:
camera_front_left:
  lidar_top/os_lidar:
    x: 0.202118
    y: -0.155210
    z: -0.504310
    roll: 1.536239
    pitch: -0.371728
    yaw: 0.015651

And as another reference my real values:
lidar_top/os_lidar:
  camera_front_left:
    x: -0.015
    y: 0.528
    z: -0.120
    roll: -1.527
    pitch: 0.0
    yaw: -0.384

knzo25, 2025-02-25:
@Sophie2216 
Inverting a transformation is a fundamental operation in math/robotics/computer vision so it falls outside the scope of this issue.
Note that other projects like x2 and xx1 perform these operations in the sensor_calibration_manager

closed 2025-02-25 by knzo25