tier4/CalibrationTools #235: The mapper can not finish the process.

sanemuytan, 2025-02-20:
Hello,

First of all thanks for sharing this repo. I'm trying to use mapping based lidar-lidar calibration tool by following the tutorial but I have some issues. I run the bag file you shared with ros2 bag play lidar_lidar.db3-001 --clock -r 0.5, after that I get those messages in the terminal.

[mapping_based_calibrator-1] [WARN] [1740033956.087515416] [calibration_mapper]: Using last pose as guess since it is lost
[mapping_based_calibrator-1] [INFO] [1740033956.113654132] [calibration_mapper]: Registrator innovation=0.00. Score=0.01
[mapping_based_calibrator-1] [INFO] [1740033956.113681517] [calibration_mapper]: Dropped frame. Unprocessed=0 Frames=52 Keyframes=21
[mapping_based_calibrator-1] [INFO] [1740033960.950867123] [mapping_based_calibrator_node]: Waiting for the mapper to finish
[mapping_based_calibrator-1] [INFO] [1740033990.956501210] [mapping_based_calibrator_node]: Waiting for the mapper to finish
[mapping_based_calibrator-1] [INFO] [1740034020.961745216] [mapping_based_calibrator_node]: Waiting for the mapper to finish

and I waited more than 30 minutes nothing has changed. Those messages continued. I have no issue to see the map built on rviz (shared an image, like you shared in the tutorial.)

According to the tutorial, 
The process of estimating transformations sends the corresponding service call with ros2 service call /stop_mapping std_srvs/srv/Empty in the terminal. Note that the user can send this command before the rosbag ends if they think the data collected is sufficient for calibration. 
In this tutorial, we send the command after the rosbag runs until the end. Once the command is sent, the displayed text should be as follows:
    [mapping_based_calibrator-1] [mapping_based_calibrator_node]: Mapper stopped through service (operator()())
    [mapping_based_calibrator-1] [calibration_mapper]: Mapping thread is exiting (mappingThreadWorker())
    [mapping_based_calibrator-1] [mapping_based_calibrator_node]: Beginning lidar calibration for pandar_front (operator()())
This process may take some time, as it involves multiple lidars' calibration. Users should remain patient and monitor the console output to track the progress of the calibration.

So, I can not see a messages when I run the command I see the output in the attachment. 

How can I handle this problem? 

Many thanks!

[image]

[image]

sanemuytan, 2025-02-20:
I played the bag file with -r 0.05, I still loose some frames. Got 11 times "Mapping failed. sensor timeout dt=1.10s" in the  terminal. The line went through the node 22.

[image]

knzo25, 2025-02-20:
`1.10s` when data is being published every `0.1s` is the problem. However, given that the problem (seems to) lies in your environment, I can no provide further support.
Is the problem is DDS, please check this:
https://autowarefoundation.github.io/autoware-documentation/main/installation/additional-settings-for-developers/network-configuration/dds-settings/

sanemuytan, 2025-02-20:
Ok, thank you. As last question, which pcl version do you use in those bags?

knzo25, 2025-02-20:
I do not really get the question, but if you are asking what PCL I use:

```bash
user@host:~/workpace/autoware$ apt list --installed | grep pcl
...
libpcl-dev/jammy,now 1.12.1+dfsg-3build1 amd64 [installed,automatic]
...
```

closed 2025-02-25 by knzo25