tier4/CalibrationTools #238: base lidar extrinsic calib using target

pedestrain123, 2025-03-11:
I want to calibrate the extrinsic between LiDAR and GNSS/INS. Can I calibrate it using the following method? Calibration principle: Place a stationary calibration board in the environment, and the laser radar point cloud extracts the coordinates of the four corner points of the calibration board in the laser radar coordinate system through an algorithm. Since the true coordinates of the four corner points of the calibration board in the world coordinate system are unknown, but the 6-degree-of-freedom pose of the combined inertial navigation system relative to the world coordinate system is known, can multiple sets of inertial navigation data and the corner point coordinates of the calibration board under the laser radar be collected at different positions, and then perform least squares optimization of the extrinsic parameters? The idea is to observe at different positions, and the position of the calibration plate in the world coordinate system remains unchanged? Another question is about data collection. Is one calibration board sufficient, or are there three calibration boards? Do you have any suggestions for this calibration plan?

knzo25, 2025-03-11:
@pedestrain123 
Hi, thanks for bringing this up.
However, we do not have experience with this type of calibration. If you want to contribute with a PR we can help with the evaluation, but otherwise, I think it is out of score for us.

knzo25, 2025-04-02:
Since there has not been any new questions, I will be closing this one

closed 2025-04-02 by knzo25