tier4/CalibrationTools #243: ground_plane_calibrator

pedestrain123, 2025-03-24:
May I ask how high the accuracy of the ground_plane_calibrator calibration method can achieve for roll and pitch calibration? Are there any requirements for data collection? Can we just collect data in a straight line? Do you have any requirements for the road surface? Is it necessary to have a flat road surface without any slope?

knzo25, 2025-03-24:
The ground plane method requires a static environment and the results will be as good as your surface (the more planar the surface, the better the results). The larger the calibration area the better (as long as it is a plane).

pedestrain123, 2025-03-24:
> 地平面方法需要静态环境，结果将与表面一样好（表面越平坦，结果越好）。校准区域越大越好（只要它是一个平面）。

What is the accuracy of your testing? How many degrees can the roll and pitch angle errors reach?

knzo25, 2025-03-24:
We do not have quantitative results due to the lack of a groundtruth that is better than this method. In our indirect tests, however, when we have a very nice planar surface the plane model fits essentially all points in a 100mx100m calibration area, with the inlier criteria being about 5cm.

A rough estimation would say that the maximum angular error would be:

```python
>>> (180 / np.pi) * np.arctan2(0.05, 50.0)
0.057295760414500616
```

knzo25, 2025-04-02:
Since there has not been any new questions, I will be closing this one

closed 2025-04-02 by knzo25