tier4/CalibrationTools #253: Marker RADAR-LiDAR Calibrator

Diogofcpereira14, 2025-05-22:
Hi @knzo25 

Im using your marker calibrator to develop extrinsic calibration between a LiDAR and a Radar. However, I have one doubt. When running the sensor_calibration_manager, after selecting my project, the calibrator and after loading the inicial Tf TREE (while running a ros2bag), i found myself with a problem: in the cmd it always shows “LidarCallback” “There were no tracks”. So, my question is, what is a track? What is my ros2bag file suppose to have? 
My lidar message type topic is PointCloud2 but i already converted the data and its structure to radar tracks. Please help me understand it or upload more doccumentation, as i find this calibrator very robust and im planning to use it in my master’s dissertation. 

Thanks in advance :)

knzo25, 2025-05-22:
@Diogofcpereira14 
Sorry, I am no longer involved with this project.
I recommend you to read the code if you want more insight

Diogofcpereira14, 2025-05-29:
Sorry then. 

One more question, how can i chance the data saved in the file where i save the calibration results? It only appears the results of the front unit base link related to the radar link. I want to show the entire tf tree. How can i do that?

@ytakano @aratafuji @moaikids @amadeuszsz @kminoda @SakodaShintaro @aohsato @vividf @YamatoAndo @muraki-t4 @takayuki5168

Diogofcpereira14, 2025-06-02:
Thanks for the response @vividf :)

Whats the difference between that branch and the one “marker_radar_lidar_calibrator_support_radars_and_transformation_algorithms”? I was using that one for now and it worked with my radar. 

One more question, how can i chance the data saved in the file where i save the calibration results? It only appears the results of the front unit base link related to the radar link. I want to show the entire tf tree. How can i do that?